Interbotix/interbotix_ros_arms #1: Planning EE pose was not successful.

nathangeorge1, 2019-11-06:
Hello,

I'm trying to use [interbotix_moveit_interface](https://github.com/Interbotix/interbotix_ros_arms/tree/master/interbotix_examples/interbotix_moveit_interface) to control the Gazebo arm.

My launch file currently looks like this...
```
<launch>

	<!-- 1. Arm -->
	<include file="$(find interbotix_descriptions)/launch/description.launch">
		<arg name="robot_name"			value="rx150" />
		<arg name="use_default_rviz"	value="false" />
	</include>

	<include file="$(find interbotix_moveit_interface)/launch/moveit_interface.launch">
		<arg name="robot_name"			value="rx150" />
		<arg name="use_actual"	value="false" />
	</include>

</launch>
```

but it's throwing this error every time I click `Plan Pose` within moveit_interface_gui
```
[N=/rx150/moveit_interface_gui,F=MoveItPlanClient-233]: Planning EE pose was not successful.
```

Am I missing a parameter?

swtrossen, 2019-11-06:
Hi Nathan,

I would change your launch file to the following:

<launch>

	<include file="$(find interbotix_moveit_interface)/launch/moveit_interface.launch">
		<arg name="robot_name"			value="rx150" />
		<arg name="use_gazebo"	    value="true" />
	</include>

</launch>

If you take a look at the diagram [here](https://github.com/Interbotix/interbotix_ros_arms/tree/master/interbotix_examples/interbotix_moveit_interface), you will see that the moveit_interface launch file actually launches the robot description for you. Also, I would change your second argument 'use_gazebo' to 'true'. 

As far as the 'Plan Pose' error message, it could be due to commanding an end-effector position that is not physically possible. Could you tell me what pose you were trying to command? When using the 'Plan Pose' button for these 5dof arms, usually some combination of 'x' and 'z' will work only. Adding in a 'y' component will make the pose physically unreachable (assuming roll, pitch, and yaw values are all 0).
For example, command the end-effector to go to an 'x' of 0.20 and a 'z' of 0.20 while keeping all others at '0' and it should work.

nathangeorge1, 2019-11-07:
I updated my launch file to this, but RViz hangs indefinitely; this is the latest message

```
[N=/rx150/controller_spawner,F=main-192]: Controller Spawner: Loaded controllers: arm_controller, gripper_controller, joint_state_controller
```

I also changed `gui` to false in interbotix_gazebo/launch/gazebo.launch, but it still hangs.

swtrossen, 2019-11-07:
Did you unpause the Gazebo physics engine? If you take a look [here](https://github.com/Interbotix/interbotix_ros_arms/tree/master/interbotix_gazebo#usage), or [here](https://github.com/Interbotix/interbotix_ros_arms/tree/master/interbotix_moveit#usage) I mentioned that when using Gazebo, it's important to call that service from the terminal. Otherwise, Rviz will hang. I see I didn't include that specifically in the 'interbotix_moveit_interface' example package, so I will update the documentation there in the 'Usage' section.

nathangeorge1, 2019-11-07:
Great that did it. I'll probably employ `launch-prefix="bash -c 'sleep 10; $0 $@'` so that the rosservice automatically gets called after gazebo has time to setup.

Turns out my end effector value was too high: was using 1m instead of .2m.

Not sure if this is a bug, though, but there appear to be 3 arms. Is there documentation for what these 3 mean? Even though I can do a successful execution using x=.2 and z=.2, the gray arm keeps moving while the green and orange arms stay still.
[image]

swtrossen, 2019-11-07:
Actually, there is! Take a look over [here](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html#step-2-play-with-the-visualized-robots). In short, green = start state, orange = goal state, gray = the planned path, and gray (another instance), the current robot pose. The documentation there tells you how you can toggle on/off the visualizations or change the colors. 

As far as why the 'orange' and 'green' states don't move in Rviz (the 'gray' ones do actually) when using the moveit_interface_gui, that has to do with the fact that there is no easy way to interface with the moveit_rviz_plugin without modifying the original source code. See more over [here](https://answers.ros.org/question/257394/how-to-change-moveit-goal-state-through-code/).

nathangeorge1, 2019-11-07:
Great that helps.

I'm still had the other problem: it seems like I can "Plan and Execute" successfully within RViz > Planning but the moveit_interface_gui can only "Plan Pose" successfully but "Execute" doesn't move the robot. 

Any ideas why the "Execute" button on moveit_interface_gui doesn't move the green robot?

nathangeorge1, 2019-11-07:
Is there another way to send a XYZ point to this system and get it to move there? I didn't see an option in RViz as it only allowed preset or random positions.

swtrossen, 2019-11-07:
> Great that helps.
>
> I'm still had the other problem: it seems like I can "Plan and Execute" successfully within RViz > Planning but the moveit_interface_gui can only "Plan Pose" successfully but [...]

As I mentioned in my earlier reply, the 'Execute' button in the moveit_interface_gui will not be able to move the 'green' or 'orange' visualizations of the robot arm since those can only be directly manipulated from the Rviz MoveIt plugin. There is currently no ROS interface (that I know of) that allows you to change the position of the 'orange' or 'green' visualizations from an external ROS node without modifying MoveIt's source code. However, the 'Execute' button should still move the gray visualization in Rviz and the Gazebo model.

> Is there another way to send a XYZ point to this system and get it to move there? I didn't see an option in RViz as it only allowed preset or random positions.

This is the main reason that I created the 'moveit_interface_gui' so that you could send custom poses to the end-effector. However, MoveIt has a C++ and Python API that allows you to send custom poses to the end-effector in code. You can see more [here](http://docs.ros.org/kinetic/api/moveit_tutorials/html/index.html#movegroup-ros-wrappers-in-c-and-python)

nathangeorge1, 2019-11-08:
That's great. I'll learn MoveIt to actuate these arms.

Thanks for the help. I can see this repo just came out, and it's a really well-documented and great codebase! I'm helping to start a new robot company; so if all goes well after we test out the Reactor series, we will hopefully buy more.

closed 2019-11-08 by nathangeorge1

swtrossen, 2019-11-08:
Sounds good! As an FYI, while MoveIt is widely known, there are other robotic manipulation packages out there that can be used to actuate the end-effector of an arm to specific XYZ and roll, pitch, yaw positions (among many other things). One of them is a package created by Northwestern called [modern_robotics](https://github.com/NxRLab/ModernRobotics/tree/master/packages/Python) which I actually use in the 'interbotix_joy_control' package. Depending on what you are trying to do with the arm, and your computer specs, it might be a good idea to do a bit of research first to find the right solution for you.

nathangeorge1, 2019-11-08:
That makes sense. I'll do some research into some ROS controllers.

nathangeorge1, 2019-11-19:
As an update, I figured out what I was looking for. I just needed to add a PlanningScene and follow this [Python tutorial](https://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html), specifically:

```
	group.set_pose_target(pose_goal)
	plan = group.go(wait=True)
```

Now I can control the interbotix arms in RViz

swinterbotix, 2019-11-19:
That's awesome! The MoveIt Python interface is pretty straightforward to use. If you ever want to work with the C++ interface, I created a small ROS API over [here](https://github.com/Interbotix/interbotix_ros_arms/blob/master/interbotix_examples/interbotix_moveit_interface/src/moveit_interface.cpp) that might come in handy. It's by no means complete, but it should be enough to get you started building your own code in C++. It's what was used behind the scenes with the moveit_interface_gui