Interbotix/interbotix_ros_arms #2: Can't connect PS3 controller

nathangeorge1, 2019-11-09:
For [interbotix_joy_control](https://github.com/Interbotix/interbotix_ros_arms/tree/master/interbotix_examples/interbotix_joy_control), I did:

- `sudo systemctl start bluetooth.service # it's on`
- `sudo bluetoothctl`
- `show`

but it shows nothing is connected. My PS3 controller is connected via USB to my Ubuntu 16.04 Virtualbox:

```
nathan@nathan-vm:~/projects/sixad-master$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 009: ID 054c:05c4 Sony Corp. DualShock 4
Bus 002 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
nathan@nathan-vm:~/projects/sixad-master$
```

I installed "sixad-master" from source and "bluez" is installed but still no luck:

```
nathan@nathan-vm:~/projects/sixad-master$ sudo sixpair
USB_REQ_GET_CONFIGURATION: Broken pipe
Current Bluetooth master: Unable to retrieve local bd_addr from `hcitool dev`.
Please enable Bluetooth or specify an address manually.
nathan@nathan-vm:~/projects/sixad-master$ hcitool dev
Devices:
nathan@nathan-vm:~/projects/sixad-master$ 
```

This [reddit post](https://www.reddit.com/r/linux_gaming/comments/5blf88/help_with_wireless_ps3_controller_on_ubuntu_1604/) leads me to think PS3 doesn't work on Ubuntu 16.04.

I got a xbox 360 controller connected. Is there a way to port the driver to xbox, or is there another way to get the PS3 controller working?

swtrossen, 2019-11-10:
Hi Nathan,

It looks like you have a DualShock 4 (PS4) controller connected based on the output of your 'lsusb' command. Unfortunately, only original Sony DualShock 3 (PS3) controllers work with the 'interbotix_joy_control' package (if you're on ROS Kinetic). If you're using the packages with ROS Melodic, it should also be possible to use off-brand PS3 controllers.

That said, there are ROS packages available that support the PS4 controller. With a little work, it should be possible to make it compatible with the 'interbotix_joy_control' package.

nathangeorge1, 2019-11-11:
I was able to use [ps4-ros](https://github.com/solbach/ps4-ros) to see ps4 output on the /joy topic.

I started it separately from the ps4.launch file, removing the current joy node:
```
<launch>

	<arg name="robot_name"                      default="rx150"/>
	<arg name="threshold"                       default="0.75"/>

	<include file="$(find interbotix_sdk)/launch/arm_run.launch">
		<arg name="robot_name"                    value="$(arg robot_name)"/>
		<arg name="arm_operating_mode"            value="velocity"/>
		<arg name="arm_profile_acceleration"      value="0"/>
		<arg name="gripper_operating_mode"        value="pwm"/>
	</include>

	<node
		pkg="interbotix_joy_control"
		type="joy_control_node"
		name="joy_control_node"
		output="screen"
		ns="$(arg robot_name)">
		<param name="threshold" value="$(arg threshold)"/>
	</node>

	<node
		pkg="interbotix_joy_control"
		type="joy_robot_control.py"
		name="joy_robot_control"
		output="screen"
		ns="$(arg robot_name)">
		<param name="robot_name" value="$(arg robot_name)"/>
	</node>

</launch>
```

Then I updated interbotix_sdk/launch/arm_run.launch:
```
  <arg name="port"                              default="/dev/input/js3"/> <!-- /dev/ttyDXL -->
```

but I still get these errors:
```
ROS_MASTER_URI=http://localhost:11311

process[rx150/robot_state_publisher-1]: started with pid [12641]
process[rx150/rviz-2]: started with pid [12642]
process[robot_transform_broadcaster-3]: started with pid [12648]
process[rx150/arm_node-4]: started with pid [12654]
process[rx150/joy_control_node-5]: started with pid [12656]
process[rx150/joy_robot_control-6]: started with pid [12664]
[N=/rx150/arm_node,F=arm_ping_motors-481]: [TxRxResult] Failed transmit instruction packet!
[N=/rx150/arm_node,F=arm_ping_motors-482]: Can't find Dynamixel ID '1'
[N=/rx150/arm_node,F=arm_ping_motors-481]: [TxRxResult] Failed transmit instruction packet!
[N=/rx150/arm_node,F=arm_ping_motors-482]: Can't find Dynamixel ID '2'
[N=/rx150/arm_node,F=arm_ping_motors-481]: [TxRxResult] Failed transmit instruction packet!
[N=/rx150/arm_node,F=arm_ping_motors-482]: Can't find Dynamixel ID '3'
[N=/rx150/arm_node,F=arm_ping_motors-481]: [TxRxResult] Failed transmit instruction packet!
[N=/rx150/arm_node,F=arm_ping_motors-482]: Can't find Dynamixel ID '4'
[N=/rx150/arm_node,F=arm_ping_motors-481]: [TxRxResult] Failed transmit instruction packet!
[N=/rx150/arm_node,F=arm_ping_motors-482]: Can't find Dynamixel ID '5'
[N=/rx150/arm_node,F=arm_ping_motors-481]: [TxRxResult] Failed transmit instruction packet!
[N=/rx150/arm_node,F=arm_ping_motors-482]: Can't find Dynamixel ID '6'
[N=/rx150/arm_node,F=arm_torque_off-371]: Arm torqued off!
[N=/rx150/arm_node,F=arm_load_motor_configs-504]: [DynamixelDriver] Failed to get the Tool
[N=/rx150/arm_node,F=arm_load_motor_configs-505]: Failed to write value[0] on items[Min_Position_Limit] to [ID : 1]
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 14) >= this->size() (which is 14)
[rx150/arm_node-4] process has died [pid 12654, exit code -11, cmd /home/nathan/catkin_ws/devel/lib/interbotix_sdk/arm_node __name:=arm_node __log:=/home/nathan/.ros/log/4bf693b2-0496-11ea-adcd-080027519de6/rx150-arm_node-4.log].
log file: /home/nathan/.ros/log/4bf693b2-0496-11ea-adcd-080027519de6/rx150-arm_node-4*.log
[rx150/joy_control_node-5] process has died [pid 12656, exit code -6, cmd /home/nathan/catkin_ws/devel/lib/interbotix_joy_control/joy_control_node __name:=joy_control_node __log:=/home/nathan/.ros/log/4bf693b2-0496-11ea-adcd-080027519de6/rx150-joy_control_node-5.log].
log file: /home/nathan/.ros/log/4bf693b2-0496-11ea-adcd-080027519de6/rx150-joy_control_node-5*.log
```

Does interbotix_joy_control work with simulations? If so, how do I use port=`/dev/input/js3` instead of `/dev/ttyDXL`?

swtrossen, 2019-11-11:
Nathan,

The 'interbotix_joy_control' package only works with the physical robot. Currently, the Gazebo simulation is only meant to be used with MoveIt (though of course, you could create your own custom node to interface with the Gazebo models). 

The 'port' argument that you are talking about describes the USB port that physically connects your computer to the actual robot arm. This should never be changed unless you connect two or more physical robots to your computer (in which case, you might change one port to '/dev/ttyUSB0' and the other to '/dev/ttyUSB1'). This 'port' is not the one that is connecting to your joystick.

As far as how your joystick gets connected to the computer, there seems to be a 'dev' parameter in the ps4.launch file that you can set to your joystick port (usually '/dev/input/js0' though it would be '/dev/input/js3' in your case).

Even after doing this, you should make sure that your joystick commands are publishing to the '/joy' topic. You might have to remap some of the buttons as well. You can do this at the top of the [joy_control_node](https://github.com/Interbotix/interbotix_ros_arms/blob/master/interbotix_examples/interbotix_joy_control/src/joy_control_node.cpp) cpp file.

nathangeorge1, 2019-11-11:
Thanks; I've been studying MoveIt and Dynamixel a little more and starting to understand.

I think what I want is more along the lines of: https://ros-planning.github.io/moveit_tutorials/doc/motion_planning_api/motion_planning_api_tutorial.html

So I'll keep researching this. Thanks

closed 2019-11-11 by nathangeorge1

swtrossen, 2019-11-11:
Sounds good, and good luck!