Interbotix/interbotix_ros_arms #3: Robot model parameter not found! Did you remap 'robot_description'?

nathangeorge1, 2019-11-19:
Hello,

Although my launch is
```
	<include file="$(find interbotix_moveit_interface)/launch/moveit_interface.launch">
		<arg name="robot_name"		value="wx200" />
		<arg name="use_gazebo"	    value="true" />
	</include>
```

and although I unpause the physics and wait a few seconds for it to finish, the following code
```
	robot = moveit_commander.RobotCommander()
```

produces this error
```
[N=/move_group_commander_wrappers_1574181380190065319,F=RDFLoader::RDFLoader-64]: Robot model parameter not found! Did you remap 'robot_description'?
```

How do I initialize the RobotCommander?

nathangeorge1, 2019-11-19:
I was able to solve this by doing:

```
	robot = moveit_commander.RobotCommander(robot_description="wx200/robot_description")
	scene = moveit_commander.PlanningSceneInterface(ns="wx200")
	group_name = "interbotix_arm" # panda_arm
	group = moveit_commander.MoveGroupCommander(robot_description="wx200/robot_description", ns="wx200", name=group_name)
```

but it led to another problem, so i'll start another issue.

closed 2019-11-19 by nathangeorge1

Seekcha, 2020-10-04:
were you able to solve the another problem ?

swinterbotix, 2020-10-05:
I believe the reason you're finding that you have to adjust the default namespace parameters in your node to 'wx200' or 'wx200/robot_description' is because you're not launching the node itself in the 'wx200' namespace. In the launch file [here](https://github.com/Interbotix/interbotix_ros_arms/blob/master/interbotix_examples/interbotix_moveit_interface/launch/moveit_interface.launch#L53), I _do_ launch the node in the robot's namespace. This way, there's no need to include it when instantiating the RobotCommander, PlanningSceneInterface, or MoveGroupCommander objects.