Interbotix/interbotix_ros_arms #4: How to set dof=6 in moveit group

avnermosh, 2019-12-27:
Hi, 

I have a rx150 and I'm testing with the python interface (using a fake robot for start)
`roslaunch interbotix_moveit_interface moveit_interface.launch robot_name:=rx150 use_fake:=true use_python_interface:=true dof:=6`

The default dof value is 5 and with it, the program runs fine
```
# ok - runs without a problem
roslaunch interbotix_moveit_interface moveit_interface.launch robot_name:=rx150 use_fake:=true use_python_interface:=true
```

The rx150 has 6 dof, so I run it like so:
```
# NOT ok - on the fake robot
roslaunch interbotix_moveit_interface moveit_interface.launch robot_name:=rx150 use_fake:=true use_python_interface:=true dof:=6

# NOT ok - on the actual robot
roslaunch interbotix_moveit_interface moveit_interface.launch robot_name:=rx150 use_actual:=true use_python_interface:=true dof:=6
```

I get an error of "list assignment index out of range"
for joint_goal because moveit group (via get_current_joint_values()) returns a list of 5 states

**How can I set the dof in moveit group to 6?**

```
# The error
('joint_goal', [0.0, 0.0, 0.0, 0.0, 0.0])
Traceback (most recent call last):
  File "/home/avnerm/interbotix_ws/src/interbotix_ros_arms/interbotix_examples/interbotix_moveit_interface/scripts/moveit_python_interface", line 642, in <module>
    main()
  File "/home/avnerm/interbotix_ws/src/interbotix_ros_arms/interbotix_examples/interbotix_moveit_interface/scripts/moveit_python_interface", line 598, in main
    step1(tutorial)
  File "/home/avnerm/interbotix_ws/src/interbotix_ros_arms/interbotix_examples/interbotix_moveit_interface/scripts/moveit_python_interface", line 430, in step1
    joint_goal[5] = pi/3
IndexError: list assignment index out of range
```

swinterbotix, 2019-12-29:
Hi avnermosh,

The rx150 robot actually only has 5dof. In other words, it only has 5 joints capable of manipulating the arm. The gripper motor is not considered part of the arm since it cannot be used to orient the end-effector - just to open and close it.

The only arms in the packages that currently have 6 dof are the 'wx250s' and 'vx300s' ('s' is for six). We do not sell them yet but they should be in our online store in the next few months. If you set the 'dof' launch file argument to '6' when using these arms, you should not see any errors.

swinterbotix, 2020-01-05:
Hi @avnermosh,

Did my comment clear up your issue? If so, I'd appreciate it if you closed it. Otherwise, let me know if there is anything else I can help you with.

avnermosh, 2020-01-05:
It did. Thanks for your help.
Avner

swinterbotix, 2020-01-05:
problem was resolved

closed 2020-01-05 by swinterbotix