Interbotix/interbotix_ros_arms #5: Problem using MoveIt to control real arm with position control

robert-ros, 2020-01-30:
Hello there!

I’m doing several tests with the real arm VX300. My setup is:

- Arm: real VX300
- Controller: OpenCM9.04-C with 485 expansion board
- Software: ROS Kinetic, Ubuntu 16.04
- Package: Interbotix ROS without modifications

I’ve managed to control the arm by velocity control successfully using MoveIt! The command that I used was:

`roslaunch interbotix_moveit interbotix_moveit.launch robot_name:=vx300 use_actual:=true dof:=5  arm_operating_mode:=velocity`

Nevertheless, when I launch the arm using position control:

`roslaunch interbotix_moveit interbotix_moveit.launch robot_name:=vx300 use_actual:=true dof:=5  arm_operating_mode:=position`

I get an error in the terminal if I move the arm using MoveIt:

`[ERROR] [1580382198.261975701]: Incorrect control mode. Please set to velocity control mode.`

Then, MoveIt moves the arm but with a strange behavior. 

[image]

So, how can I set to position control mode? 

Thanks a lot fellows!

swinterbotix, 2020-01-30:
Hi Robert,

I discovered and fixed this bug a few days ago. If you pull the latest code from the repo, you should no longer see that error, and position control should work just fine.

robert-ros, 2020-01-31:
Now it works!

Thank you :)

closed 2020-01-31 by robert-ros