Interbotix/interbotix_ros_arms #6: Problem using PS3 and PS4: joy_robot_control.py dies

robert-ros, 2020-01-31:
Hello again!

I’m testing interbotix_joy_control package . My setup is:
- Arm: real VX300 
- Controller: OpenCM9.04-C with 485 expansion board 
- Software: ROS Kinetic, Ubuntu 16.04 
- Package: interbotix_ros_arms (updated on 31/1/2020)
- Joystick: PS3 original and PS4 original

When I launch the package interbotix_joy_control using PS3 or PS4 controller, the script joy_robot_control.py dies. 

With PS3:

`roslaunch interbotix_joy_control joy_control.launch robot_name:=vx300 controller:=ps3`

With PS4:

`roslaunch interbotix_joy_control joy_control.launch robot_name:=vx300`

In both cases I get the following error:

`[vx300/joy_robot_control-8] process has died [pid 21275, exit code 1, cmd /home/robert/interbotix_ws/src/interbotix_ros_arms/interbotix_examples/interbotix_joy_control/scripts/joy_robot_control.py __name:=joy_robot_control __log:=/home/robert/.ros/log/9450da28-440b-11ea-8493-58fb84519d84/vx300-joy_robot_control-8.log].log file: /home/robert/.ros/log/9450da28-440b-11ea-8493-58fb84519d84/vx300-joy_robot_control-8*.log `

[image]

I have followed the steps indicated in the readme of the interbotix_joy_control package to connect the joystick. 

I think that the connection is ok because I checked the state of the joystick buttons using jstest

` jstest /dev/input/js0`

Terminal shows:

[image]

Moreover I have checked the PS3 controller with ps3joy ROS package and it works.

Any suggestions mates?

swinterbotix, 2020-01-31:
It looks like you haven't yet installed the Modern Robotics library that the joy_control package depends on. The instructions to do this step were outlined in the Quickstart section but essentially, you need to do the following in a terminal.

$ sudo apt install python-pip
$ sudo pip install modern_robotics

And you should be good to go!

robert-ros, 2020-02-03:
It works!

It was a problem with python in my machine . I installed the package when I started to work with the arm. Nevertheless I installed it using  pip3 because with pip I got this error:

```
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    from pip import main
ImportError: cannot import name main
```

For someone else with this problem, I just solved it by upgrading pip 

`pip install --user --upgrade pip`

If I do:

` sudo pip install modern_robotics`

Now I can install it and interbotix_joy_control package works 

Thank you :)

closed 2020-02-03 by robert-ros