Interbotix/interbotix_ros_arms #10: Integrating the arm with a mobile platform

bjsowa, 2020-02-16:
Hello!

I am in the process of integrating the WidowX 250 arm with a robotic platform and I encountered some problems.

My robot uses these files to start the basic functionality:
  - **robot.urdf.xacro** - the URDF model of the robot
  - **robot.launch** - uploads the URDF model to the `robot_description` parameter and starts some nodes (including `robot_state_publisher`)

I wanted to extend the model by the URDF of the arm and start the arm controller nodes, so I added this to the `robot.urdf.xacro` file:

```XML
<xacro:arg name="robot_name" default="wx250"/>
<xacro:include filename="$(find interbotix_descriptions)/urdf/$(arg robot_name).urdf.xacro"/>

<joint name="arm_joint" type="fixed">
  <origin xyz="0.043 0 0"/>
  <parent link="base_link"/>
  <child link="$(arg robot_name)/base_link"/>
</joint>
```
and this to the `robot.launch` file:
```XML
<include file="$(find interbotix_sdk)/launch/arm_run.launch">
    <arg name="robot_name" value="wx250"/>
    <arg name="use_default_rviz" value="false"/>
</include>
```
And this seems to work. I can visualize the arm model attached to the robot in RViz and the topics and services for joint commands work. 

Now, I wanted to add moveit functionality, so I set the `use_moveit` parameter to `true`:
```XML
<include file="$(find interbotix_sdk)/launch/arm_run.launch">
    <arg name="robot_name" value="wx250"/>
    <arg name="use_moveit" value="true"/>
    <arg name="use_default_rviz" value="false"/>
</include>
```
but this results in the arm's base link being attached to the `world` frame (due to arm's state publisher reading different `robot_description`), so the TF tree now has 2 root links.

I don't want to modify the packages without understanding the full concept, so I wanted to ask. Is there something I am doing wrong or the thing I am trying to do is not possible with the current state of the packages?

swinterbotix, 2020-02-16:
Hi @bjsowa,

It should not be necessary to include the URDF of the WX250 robot in your robot's URDF.xacro file as the interbotix_sdk package will load it for you automatically (you would still have to create the 'arm_joint' though). As you suspected, there will be two robot description parameters. One called 'robot_description' that will be hosting your mobile base, and then another called /wx250/robot_description for the interbotix arm. Setting the 'use_moveit' argument will result in two things happening. 1) the sdk node will start the follow joint trajectory action servers that MoveIt expects to see, and 2) the wx250.urdf.xacro file has a 'moveit' parameter that will create a joint between the world frame and the arm's base link frame. There are two reasons why the second thing happens - 1) if both MoveIt and Gazebo are being used, Gazebo will not fix the arm's base link to its own world frame unless the connection is specifically made in the URDF. Without the connection, the arm will just faceplant in Gazebo due to gravity. The second reason is so that no matter what arm you are using, all of them will share a common world frame so that if you are using Rviz with MoveIt, you don't have to keep changing the 'fixed frame' to /<robot_name>/base_link depending on what robot you have.

Since both of those reasons don't really apply to you (you are fixing the robot arm to a mobile base),  you could just go to the wx250.urdf.xacro file and comment out the xacro-if condition block that checks to see if MoveIt is being used and if yes, connects the world and base link frames (lines 20-26).

Let me know if that works for you!

Best,
Solomon

swinterbotix, 2020-02-18:
Just as an FYI, I plan to release an update soon that will open up a launch file argument allowing you to choose whether or not `/<robot_name>/base_link` will be connected to 'world'.

swinterbotix, 2020-02-19:
Update addressing this issue was released.

bjsowa, 2020-02-21:
Sorry for the late replay and thanks for the update. I was going to create a PR which adds a similar option, but I also don't think there is a need for using the `world` frame in moveit. 

> if both MoveIt and Gazebo are being used, Gazebo will not fix the arm's base link to its own world frame unless the connection is specifically made in the URDF. Without the connection, the arm will just faceplant in Gazebo due to gravity. 

This is a valid reason to fix the arm's base_link in Gazebo

> The second reason is so that no matter what arm you are using, all of them will share a common world frame so that if you are using Rviz with MoveIt, you don't have to keep changing the 'fixed frame' to /<robot_name>/base_link depending on what robot you have.

I understand that you want a common RViz config for all arms, but I don't know if you realized that the fixed frame for RViz can be passed through a command arg. Please take a look at #11 which removes the use of `world` frame and `fixed` joint in moveit.

There are 2 reasons why I want to include the URDF of the arm into the URDF of the robot:
1. Viewing the robot's model with all the attached add-ons from a single `RobotModel` display in RViz: 
[image]
2. making MoveIt aware of the robot's collision model when performing trajectory planning. This would require replacing the URDF and SRDF models read by MoveIt (`robot_name/robot_description` and `robot_name/robot_description_semantic`) and I have not came up with the idea of how to add such functionality yet.

swinterbotix, 2020-02-21:
> This is a valid reason to fix the arm's base_link in Gazebo

Are you saying that you know a way to do this without using a world frame? I'm all ears if you do! Also regarding your PR, if you don't use a world frame, MoveIt get's mad that the 'root link' of the URDF contains inertial values. It's not an issue if you're placing the arm on a mobile platform, but it is an issue if you are using the arm standalone. If you know of a fix for this (that does not involve creating a dummy link - since that just feels like bad design) that will work regardless if MoveIt is used in Gazebo (without faceplanting) or on the physical robot, I'd appreciate it! 

> I understand that you want a common RViz config for all arms, but I don't know if you realized that the fixed frame for RViz can be passed through a command arg. Please take a look at #11 which removes the use of world frame and fixed joint in moveit.

I did not know that this was possible. Thanks for bringing it to my attention. 

> There are 2 reasons why I want to include the URDF of the arm into the URDF of the robot:

Currently, I'm designing 4 mobile platforms complete with Interbotix Arms. The way I've worked around your first issue about having a single `robot_description` parameter is to include the arm's urdf.xacro file into your own URDF file, and load the whole thing to `/<robot_name>/robot_description` (note that you would have to edit line 15 [here](https://github.com/Interbotix/interbotix_ros_arms/blob/master/interbotix_descriptions/launch/description.launch#L15) so that instead of loading `<robot_model>.urdf.xacro`, you load `<your_robo_urdf>.urdf.xacro` file). Normally, when using a standalone Interbotix arm, the `robot_name` and `robot_model` launch file arguments would be equivalent. However, if using a mobile platform, you can specify `robot_name` as `mobile_wx250` and `robot_model` as `wx250`. Then your URDF will be launched in `/mobile_wx250/robot_description`, and the interbotix_sdk will still know what arm you are using from the `robot_model` argument. Additionally, the interbotix_sdk's `get_robot_info` service parses the `/<robot_name>/robot_description` parameter so that it can get joint limit info - so the Interbotix arm's URDF must be in that parameter.

About the second issue...

Once you have your overall robot in `mobile_wx250/robot_description`, that's what MoveIt will see as well. So to get the SRDF up to date, place your custom SRDF file (look at the last paragraph to see how to make it) in the interbotix_movet/config/srdf directory and call it something meaningful (like 'mobile_wx250'). Then in the [planning_context](https://github.com/Interbotix/interbotix_ros_arms/blob/master/interbotix_moveit/launch/planning_context.launch#L15) launch file (line #15), change `$(arg robot_name).srdf.xacro` to `mobile_wx250.srdf.xacro`. 

Unfortunately, I did not include a `robot_model` argument/parameter in the interbotix_moveit package as at the time (six months ago or so), I was only designing for the arms to be used as standalone robots. Ideally though, you could then launch the interbotix_moveit.launch file, setting `robot_name:=mobile_wx250`, and `robot_model:=wx250`. (I'll try to work on that soon. From this conversation, it also looks like adding a third argument like `robot_file` would be beneficial. `robot_file` would default to `robot_model` but could be changed to point to a custom URDF or SRDF file so the modifications I discussed in the paragraph above wouldn't be necessary).  

Regarding generating the custom SRDF, the best way to do that is run the MoveIt Setup Assistant Wizard on your full robot to generate a raw 'moveit' package (nothing fancy - just run the 'self-collisions' tab, add 'author information' go to 'Configuration Files', choose a temp directory and generate the files) . Next, create a duplicate of the 'wx250.srdf.xacro' file located in the interbotix_moveit/config/srdf directory and call it 'mobile_wx250.urdf.xacro'. Then all you have to do is compare the SRDF you generated with the duplicate and add the right 'disable collision' tag info (between the arm and your base) to the duplicate file (also change the robot name in the SRDF to the robot name you used in your custom URDF). 

I think that should do it. Before I do any modifications though, I'd appreciate hearing your answers to my questions and thoughts on my approach.

bjsowa, 2020-02-21:
> Are you saying that you know a way to do this without using a world frame? I'm all ears if you do! Also regarding your PR, if you don't use a world frame, MoveIt get's mad that the 'root link' of the URDF contains inertial values.

I was saying that arm's base link should be fixed to the world frame in Gazebo, but does not have to be fixed on the real robot which I showed in #11. 

About the warning about the root link containing inertial values, I might not be 100% sure but I think it is only an issue, when KDL is used to calculate things related to robot's dynamics. As far as I know, MoveIt does not use these functionalities and the warning can be ignored.

bjsowa, 2020-02-21:
I don't have any issues with using moveit even though the KDL warning shows. I was even able to use the KDL kinematics solver without any problems

swinterbotix, 2020-02-21:
> I was saying that arm's base link should be fixed to the world frame in Gazebo, but does not have to be fixed on the real robot which I showed in #11.

Ok, I'm not sure I like that approach since I feel that the setup should be consistent regardless if you are using a MoveIt generated fake robot, a Gazebo simulated real robot, or the actual physical robot. By including an optional world frame that attaches to the base link in the URDF, that gives you the freedom to toggle it on/off for all three setups in one shot. Also, I understand that the KDL warning doesn't cause any issues, but as the world frame will take care of that issue as well, it's another small reason to keep it.

bjsowa, 2020-02-21:
Ok, I restored the use_world_frame parameter in #11 to the way it was before. Could you give feedback in the Pull Request now?

bjsowa, 2020-03-01:
I noticed that all of the examples in `interborix_examples` directory start the arm driver node. This makes it impossible to use the packages as they are with a remotely controlled arm (like in my case). I would appreciate if you can add an option to not include the `arm_run.launch` file.

swinterbotix, 2020-03-01:
@bjsowa,

Can you explain what you mean? If your robot computer contains the interbotix_ros_arm repo, there's nothing stopping you from SSHing into that computer from a remote computer and starting the example packages.

Are you saying that since in your case, you have custom settings for the driver node, and as the example packages don't allow full customization for the driver node, you can't run them easily?

bjsowa, 2020-03-01:
I have a robot that starts the arm driver (`arm_run.launch`) and a computer which is configured to be visible in the ROS network. I want to run the examples from my computer. All of the example launch files include the `arm_run.launch` file which tries to run the driver. That results in failure because the arm is not connected to the computer but is remotely controlled. 

A solution I propose is to add an argument to the launch files that would let the user choose whether to run the driver (include the `arm_run.launch`). The `interbotix_moveit.launch` allows such option by leaving all `use` parameters to false.

swinterbotix, 2020-03-02:
I see.

The reason the interbotix_moveit.launch file is structured the way it is, is because there are three ways one can control the arm there. If the 'use_actual' argument was always set to 'true', a user would always have to set it to false before setting another control method (gazebo or fake) to 'true' if that's what the user wanted to do - which isn't great from a design perspective. 

Regarding the interbotix_example packages, there's only one way for the packages to run, and that's via the `arm_run.launch` file. As such, I didn't feel the need for there to be a 'use_actual' argument. Additionally, each example package has slightly different configurations for the `arm_node` parameters. If the `arm_run.launch` file was made optional in these scenarios, it would be up to the user to correctly set the `arm_node` parameters on the robot for a specific example package and correctly load the `robot_description` parameter - which seems a bit of a hassle. In other words, there's no magic configuration of the `arm_run.launch` file that would allow all the example packages to work.

On a different note, the 'interbotix_moveit_interface' package and the 'python_demo' scripts already can run on a remote computer while the `arm_run.launch` file runs on the robot itself.

bjsowa, 2020-03-10:
> The reason the interbotix_moveit.launch file is structured the way it is, is because there are three ways one can control the arm there. If the 'use_actual' argument was always set to 'true', a user [...]

I understand the motivation behind the parameters and also think this is a good design. I am just saying that it allows, intentionally or not, to use the driver that is already running on the ROS network (either real or simulated).

> each example package has slightly different configurations for the arm_node parameters. If the arm_run.launch file was made optional in these scenarios, it would be up to the user to correctly set [...]

I don't see a problem with it. Do you perhaps see a better solution?

> On a different note, the 'interbotix_moveit_interface' package and the 'python_demo' scripts already can run on a remote computer while the arm_run.launch file runs on the robot itself.

I tested the python scripts and they worked as expected which is nice. But I also think that at least the `joy_control.launch` file should give the user the option to not run the driver.

My proposition:
```xml
  <arg name="run_arm"                         default="true"/>

  <include if="$(arg run_arm)" file="$(find interbotix_sdk)/launch/arm_run.launch">
```

swinterbotix, 2020-03-18:
Hi @bjsowa,

I've implemented an 'arm_run' launch file argument in the interbotix_examples ROS packages that addresses this issue. Feel free to take a look!

bjsowa, 2020-03-20:
Thanks, that's all I have for now. I'm closing the issue.

closed 2020-03-20 by bjsowa