Interbotix/interbotix_ros_arms #16: Gazebo & RVIZ != love

Ktysai, 2020-03-26:
Hi there,

I've began again to play with ROS and I want to implement something with PX100.

I've put on top of it a Pixy camera for image recognition, but it seems I'm putting at risk the robot and the camera with some risky moves.

So... I wanted to make the code tested in Gazebo and to have Rviz to show the camera vision. I've modified the urdf & gazebo files.

Gazebo & RVIZ does work separate, but if I start Gazebo and then Rviz, it's a problem. Rviz will show incorrectly the robot position, the camera is black, but most important is presenting issues at Robot model, just like it's a TF problem.

Any suggestions?

Stay Strong!
Ktysai

swinterbotix, 2020-03-27:
Hi @Ktysai,

Did you make sure to unpause Gazebo's physics engine? By default, I've configured Gazebo to start in a paused state to give time for the ROS controllers to kick in. As a result, Gazebo will not start publishing to `/tf` or the `/joint_states` topics until it's unpaused. To do that, you should type `rosservice call /gazebo/unpause_physics` in a terminal.

So to get Gazebo and Rviz to play nice, you would do...
`roslaunch interbotix_gazebo gazebo.launch robot_name:=px100 use_default_rviz:=true`

Then open another terminal and type...
`rosservice call /gazebo/unpause_physics`

Note that currently, the only way to interface with the arm joints in Gazebo is to create an Action Client that sends FollowJointTrajectory messages as described [here](http://wiki.ros.org/joint_trajectory_controller) or publish trajectory messages to the arm_controller/command topic. At some point, I would like to create a node that exposes the `joint/commands` and `single_joint/command` topics to the user (just like by the real robot), processes the data, and then publishes commands directly to the right ROS Control topics in Gazebo (of course, I would need to add effort_controllers/JointPositionControllers and/or effort_controllers/JointVelocityControllers to do this).

Hope that helps!

Ktysai, 2020-03-27:
@swinterbotix 

Thanks for the elaborate answer! 
I've used the used the unpaused command rather random I might say.

Let me test it bit by bit. It's great opportunity to learn.

Stay Strong!
Ktysai

Ktysai, 2020-03-28:
@swinterbotix please intervene if something is not correct!

For other having an Interbotix arm and wants to play :)

Short tutorial on how to have in RVIZ a camera vision, when the simulation runs in GAZEBO.
This case is for robot arm PX100

1. `cd ~/interbotix_ws/src/interbotix_ros_arms/interbotix_descriptions/urdf`
2. Make a copy of the file `px100.urdf.xacro`
3. With your preferate editor editor add

`    

    <joint name="camera_joint" type="fixed">
      <axis xyz="0 1 0" />
      <origin xyz="0 0 0.070" rpy="0 0 0"/> 
      <parent link="$(arg robot_name)/gripper_bar_link"/>
      <child link="$(arg robot_name)/camera_link"/>
      
    </joint>

    
    <link name="$(arg robot_name)/camera_link">
      <collision>
        <origin xyz="0 0 0" rpy="0 0 0"/>
        <geometry>
      <box size="${camera_link} ${camera_link} ${camera_link}"/>
        </geometry>
      </collision>

      <visual>
        <origin xyz="0 0 0" rpy="0 0 0"/>
        <geometry>
      <box size="${camera_link} ${camera_link} ${camera_link}"/>
        </geometry>
        <material name="red"/>
      </visual>

      <inertial>
        <mass value="1e-5" />
        <origin xyz="0 0 0" rpy="0 0 0"/>
        <inertia ixx="1e-6" ixy="0" ixz="0" iyy="1e-6" iyz="0" izz="1e-6" />
      </inertial>
    </link>

    `

4. Afterwards go to and make a backup copy and open the file a `/home/vio/interbotix_ws/src/interbotix_ros_arms/interbotix_gazebo/config/interbotix_texture.gazebo`

Ktysai, 2020-03-28:
5. 
`  
  <gazebo reference="$(arg robot_name)/camera_link"> 
    <sensor type="camera" name="camera1">
      <update_rate>30.0</update_rate>
      <camera name="head">
        <horizontal_fov>1.3962634</horizontal_fov>
        <image>
          <width>640</width>
          <height>400</height>
          <format>R8G8B8</format>
        </image>
        <clip>
          <near>0.02</near>
          <far>300</far>
        </clip>
        <noise>
          <type>gaussian</type>
          
          <mean>0.0</mean>
          <stddev>0.007</stddev>
        </noise>
      </camera>
      <plugin name="camera_controller" filename="libgazebo_ros_camera.so">
        <alwaysOn>true</alwaysOn>
        <updateRate>0.0</updateRate>
        <cameraName>camera1</cameraName>
        <imageTopicName>image_raw</imageTopicName>
        <cameraInfoTopicName>camera_info</cameraInfoTopicName>
        <frameName>camera_link</frameName>
        <hackBaseline>0.07</hackBaseline>
        <distortionK1>0.0</distortionK1>
        <distortionK2>0.0</distortionK2>
        <distortionK3>0.0</distortionK3>
        <distortionT1>0.0</distortionT1>
        <distortionT2>0.0</distortionT2>
      </plugin>
    </sensor>
  </gazebo>`
6. `cd ~/interbotix_ws`
`source devel/setup.bash`
7. The commands @swinterbotix recommends
So to get Gazebo and Rviz to play nice, you would do...
`roslaunch interbotix_gazebo gazebo.launch robot_name:=px100 use_default_rviz:=true`

Then open another terminal and type...
`rosservice call /gazebo/unpause_physics`

The point "5."  I'm not sure why it's so bad formatted.

Hope it helps others

swinterbotix, 2020-03-29:
Awesome! Looks good to me.

swinterbotix, 2020-04-08:
Closing since issue was resolved. Feel free to reopen if you have other questions!

closed 2020-04-08 by swinterbotix