Interbotix/interbotix_ros_arms #19: px150 wrist_rotate joint limits cause cable to be unplugged

TaekedeHaan, 2020-05-19:
With the current joint limits (`lower="${-pi}" upper="${pi}"`) set [here](https://github.com/Interbotix/interbotix_ros_arms/blob/731e0553bda22e9713bb648da43a008487500fb5/interbotix_descriptions/urdf/px150.urdf.xacro#L208) my wrist joint cable got unplugged while operating the px150. This can be prevented by lowering the joint limits. For me `lower="${radians(-110)}" upper="${radians(110)}"`degree seems to work.

swinterbotix, 2020-05-19:
Hi @TaekedeHaan,

The 3pin cable connecting the wrist_rotate motor to the wrist_angle motor should be long enough to allow the wrist_rotate joint to move between -pi and pi radians. If your cable is too short, just shoot an email to trsupport@trossenrobotics.com, and they should be able to help you out.

TaekedeHaan, 2020-05-20:
Okay thanks, for now it is no problem. I was just afraid I broke something when I heard a snapping sound.

closed 2020-05-20 by TaekedeHaan