Interbotix/interbotix_ros_arms #20: How to run this package on ROS noetic.

m2kulkarni, 2020-05-28:
What changes do i have to make to run this package on ros noetic. From what i can see, kinetic and melodic branches don't have that much difference. Any specific changes to make to any file?

swinterbotix, 2020-05-28:
Haven't looked into this so much. My guess is that as long as rosdep can find dependency packages in Noetic, you're good to go. However, considering how Noetic was just released, most likely you'll have to build a lot of dependency packages from source as they've probably not been ported over yet. I know for sure that the [dynamixel_workbench_toolbox](https://github.com/ROBOTIS-GIT/dynamixel-workbench/tree/master/dynamixel_workbench_toolbox) package and the [DynamixelSDK](https://github.com/ROBOTIS-GIT/DynamixelSDK) package aren't officially available on Noetic - both of which are dependencies for the ROS packages in this repository (though they should probably build successfully from source). I don't believe MoveIt, joy, trac_ik, etc... are ported over yet either. So unless you're willing to build a lot of packages from source (and there's no guarantee you won't run into issues while building _those_ packages from source, I'd recommend waiting a bit.

m2kulkarni, 2020-05-28:
I may try building all of them from source. I just need interbotix_gazebo and interbotix_description packages for my work. 

Are other packages required for running only gazebo & description packages?

I did catkin_make and it was built successfully. Then i sourced devel/setup.bash but still i cannot see interbotix_description & interbotix_gazebo when i try roslaunch. Any idea why that may happen?

swinterbotix, 2020-05-28:
I just checked the dependencies for the interbotix_gazebo and interbotix_descriptions packages and it seems that all the dependency packages are currently supported in Noetic. So, performing the rosdep and catkin_make commands in the QuickStart should work (though you will either have to blacklist or delete the other packages in the repository to prevent rosdep and catkin_make from trying to install their dependencies or build them respectively).

About your issue, can you type `env | grep ROS` in your terminal after sourcing your workspace and paste the output here? It could be that you sourced your workspace in one terminal, then opened another and tried to run the roslaunch command. However, that will not work since you never sourced your workspace in the second terminal.

If that's not the case - I have run into a strange behavior before where after building and sourcing a workspace, tab completion didn't work the first time trying to launch a launch file. However, the launch command still worked and tab completion worked every time afterwards.

m2kulkarni, 2020-05-28:
I guess as you said, catkin_make and rosdep had conflicts with other packages. I ran this in another workspace and it worked.

Regarding the third point, i too have had this issue in the past, dont know what causes this weird behaviour. 

Anyway thanks for the quick replies

swinterbotix, 2020-05-28:
No problem!

closed 2020-05-28 by swinterbotix