Interbotix/interbotix_ros_arms #21: Moveit does not wait for trajectory completion

technoprateek, 2020-06-03:
I have a custom robot arm using dynamixel motor for which I have successfully made the urdf according to the requirements and it works with this package.
I am able to control the real robot as well as in gazebo using the interbotix functions.

The issue is that when I send multiple points together to for execution,
for example point a,b,c,d on real robot it start moving towards b but immediately changes course and moves directly to d.
If I give only 2 points it works perfectly.
In gazebo multiple point execution works perfectly without any issue so looks like the issue is with the driver. The driver seems to accept the goal and return completed/succeeded immediately before even the trajectory is completed.

 while row is not None:
>                     joint_goal = self.group.get_current_joint_values()
>                     joint_goal[0] = row[0]
>                     joint_goal[1] = row[1]
>                     joint_goal[2] = row[2]
>                     joint_goal[3] = row[3]
>                     joint_goal[4] = row[4]
>                     joint_goal[5] = row[5]
>                     rospy.logwarn("sent goal to moveit")
>                     self.group.go(joint_goal, wait=True)
>                     rospy.logwarn("goal reached sending
> next goal if available")
>                     self.group.stop()

self.group.go(joint_goal, wait=True) This line of the code is completed immediately even before the trajectory execution is finished.

We are using gears for the motors so to account for them 
I divide the position by "gear_ratio"
In lines  1385 and 1434 I divide by the gear ratio
position = dxl_wb.convertValue2Radian(joint.motor_id, get_position[index]);
position = position/gear_ratio;

and multiply by "gear_ratio" before sending the joint values to the motors in 
void RobotArm::arm_set_single_joint_angular_position and void RobotArm::arm_write_joint_commands.

The issue of returning completed trajectory/success before reaching it is due to the something the way the driver works or there is something wrong in the way I am implementing the gear ratio.
Rest of functions though seems to work fine. Like single joint movement, multi joint movement and even just moving between only 2 points.

swinterbotix, 2020-06-04:
Can you post a snapshot of your terminal screen where these messages appear?

technoprateek, 2020-06-04:
This is the output.
It shows up warnings thats the joint value is not as expected. (which is exactly 6.7 times which is the gear_ratio i mentioned)
I am trying to have moveit is getting and sending out commands, within it's normal range of operation as I am dividing joint_states values before it is published and multiply the joint cmds before sending those commands to the motors.
In the example below i send point a to b to c to d.
In the terminal output below, Completed trajectory execution with status SUCCEEDED is shown twice even before robot moves to point b or c and it finally just moves directly to d from a.

```
[ INFO] [1591280519.364279682]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1591280519.364705220]: Planning attempt 1 of at most 1
[New Thread 0x7fff97fff700 (LWP 13307)]
[ INFO] [1591280519.370735383]: Planner configuration 'interbotix_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1591280519.372606440]: RRTConnect: Starting planning with 1 states already in datastructure
[Thread 0x7fff97fff700 (LWP 13307) exited]
[ INFO] [1591280519.388024159]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1591280519.388176096]: Solution found in 0.015883 seconds
[ INFO] [1591280519.398292129]: SimpleSetup: Path simplification took 0.009755 seconds and changed from 3 to 2 states
[New Thread 0x7fff97fff700 (LWP 13308)]
[New Thread 0x7fff977fe700 (LWP 13309)]
[ WARN] [1591280519.408942356]: shoulder motor is not at the correct initial state.
[ WARN] [1591280519.409075140]: Expected state: 1.770214, Actual State: 0.264440.
[ WARN] [1591280519.409169506]: elbow motor is not at the correct initial state.
[ WARN] [1591280519.409231211]: Expected state: 0.363553, Actual State: 0.054262.
[ WARN] [1591280519.409341539]: forearm_roll motor is not at the correct initial state.
[ WARN] [1591280519.409421776]: Expected state: 0.131922, Actual State: 0.019461.
[ INFO] [1591280519.409506645]: Succeeded to get joint trajectory!
[ INFO] [1591280519.903753303]: Trajectory done being executed.
[ INFO] [1591280519.910274969]: Controller arm_controller successfully finished
[ INFO] [1591280519.934144223]: Completed trajectory execution with status SUCCEEDED ...
[Thread 0x7fff977fe700 (LWP 13309) exited]
[Thread 0x7fff97fff700 (LWP 13308) exited]
[ INFO] [1591280520.001989464]: Received event 'stop'
[ INFO] [1591280520.013729859]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1591280520.013792496]: Planning attempt 1 of at most 1
[New Thread 0x7fff977fe700 (LWP 13310)]
[ INFO] [1591280520.014387601]: Planner configuration 'interbotix_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1591280520.014504565]: RRTConnect: Starting planning with 1 states already in datastructure
[Thread 0x7fff977fe700 (LWP 13310) exited]
[ INFO] [1591280520.026005203]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1591280520.026109873]: Solution found in 0.011625 seconds
[ INFO] [1591280520.031688533]: SimpleSetup: Path simplification took 0.005485 seconds and changed from 3 to 2 states
[New Thread 0x7fff977fe700 (LWP 13311)]
[Thread 0x7fff977fe700 (LWP 13311) exited]
[New Thread 0x7fff97fff700 (LWP 13312)]
[ WARN] [1591280520.035631650]: shoulder motor is not at the correct initial state.
[ WARN] [1591280520.035727525]: Expected state: 1.250194, Actual State: 0.181559.
[ WARN] [1591280520.035770492]: forearm_roll motor is not at the correct initial state.
[ WARN] [1591280520.035819940]: Expected state: -0.165670, Actual State: -0.025872.
[ INFO] [1591280520.035866372]: Succeeded to get joint trajectory!
[ INFO] [1591280520.663703954]: Trajectory done being executed.
[ INFO] [1591280520.666372116]: Controller arm_controller successfully finished
[ INFO] [1591280520.696140495]: Completed trajectory execution with status SUCCEEDED ...
[Thread 0x7fff97fff700 (LWP 13312) exited]
[ INFO] [1591280520.710401710]: Received event 'stop'
[ INFO] [1591280520.733649680]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1591280520.733732431]: Planning attempt 1 of at most 1
[New Thread 0x7fff97fff700 (LWP 13313)]
[ INFO] [1591280520.734190076]: Planner configuration 'interbotix_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1591280520.734335109]: RRTConnect: Starting planning with 1 states already in datastructure
[Thread 0x7fff97fff700 (LWP 13313) exited]
[ INFO] [1591280520.744911593]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1591280520.744944010]: Solution found in 0.010654 seconds
[ INFO] [1591280520.748457603]: SimpleSetup: Path simplification took 0.003480 seconds and changed from 4 to 2 states
[New Thread 0x7fff97fff700 (LWP 13314)]
[New Thread 0x7fff977fe700 (LWP 13315)]
[ WARN] [1591280520.753956239]: shoulder motor is not at the correct initial state.
[ WARN] [1591280520.753990786]: Expected state: 0.645806, Actual State: 0.092497.
[ WARN] [1591280520.754010118]: elbow motor is not at the correct initial state.
[ WARN] [1591280520.754024956]: Expected state: 0.690291, Actual State: 0.109210.
[... cut ...]
[ INFO] [1591280521.989366109]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1591280521.989485649]: Solution found in 0.013692 seconds
[ INFO] [1591280521.995057855]: SimpleSetup: Path simplification took 0.005461 seconds and changed from 3 to 2 states
[New Thread 0x7fff977fe700 (LWP 13317)]
[Thread 0x7fff977fe700 (LWP 13317) exited]
[ERROR] [1591280522.004116978]: 
Invalid Trajectory: start point deviates from current robot state more than 0.01
joint 'elbow': expected: 0.448289, current: 0.46111
[ INFO] [1591280522.008632371]: Received event 'stop'
```

swinterbotix, 2020-06-04:
i assume that the gear ratio you are talking about means that for every 6.7 revolutions of the Dynamixel Servo (the driver gear), the outer (or driven) gear rotates once.

Since you're accounting for the gear ratio in the driver node, this means that you should be sending commands from MoveIt in terms of what the joint positions should be after the gear reduction (ex. if the desired position of the shoulder joint is pi/2 radians, then you should be sending pi/2 radians, at which point the driver will take that, multiply it by 6.7 and command the motor). In this manner, MoveIt receives and commands joint values post-gear-reduction. I'm not sure why, but it looks like you are commanding the joints from MoveIt in terms of joint values pre-gear-reduction (i.e. what the Dynamixel servo positions should be and not what the _driven_ gear should be).

As a side note, if the Dynamixel Servos are supposed to be able to rotate up to +/- 6.7 revolutions, you should be setting the arm_operating_mode to 'ext_position'. By default, the motors are set to 'position' control mode and will ignore any commands trying to make them rotate past +/- 1 revolution (this is just how the firmware in the motors was configured for safety reasons I expect).

technoprateek, 2020-06-05:
The driver here did not support ext_position implemented even though there were some mentions of it. I made changes in the driver so it works with ext_position. And it works when I manually send the commands to the joints even above +/- 1.

I am trying to do exactly what you are suggesting.

in the function`
void RobotArm::arm_update_joint_states(const ros::TimerEvent &e)`

I divide the position by "gear_ratio" So that moveit does receives the the joint position after division.
So in the following line the position values will be for example 6.7(ext_position) as it is coming from the motors. I divide it by 6.7 so that moveit always gets it between +/-1(I understand these are in radians so -2pi to 2pi)

```
In lines 1385 and 1434 I divide by the gear ratio
position = dxl_wb.convertValue2Radian(joint.motor_id, get_position[index]);
position = position/gear_ratio;

```
As this the function which publishes to joint_states. I checked joint_states topic and its values are between -2pi to 2pi. But the warning says that moveit expects it to be different. That's what I am not able to understand.

**I think this is where the problem lies. There maybe somewhere else from where moveit gets the joint states and I am missing that**  

I also take the output of moveit and multiply it 6.7 to send to motors.(This second part works perfectly)
multiply by "gear_ratio" before sending the joint values to the motors in the functions
```
void RobotArm::arm_set_single_joint_angular_position 
void RobotArm::arm_write_joint_commands.

```

swinterbotix, 2020-06-07:
> The driver here did not support ext_position implemented even though there were some mentions of it. (Please correct me if I overlooked the implementation somehow but from what I found that it was not implemented fully) I made changes in the driver so it works with ext_position. And it works when I manually send the commands to the joints even above +/- 1.

I took another look, and you're right! I decided against implementing ext_position control for the arm joints since it wasn't necessary for the robot arms we develop. As an FYI though, I did a complete redesign of the interbotix_sdk that I plan to release in the next couple months. In the new version, it _does_ allow you to set a group of joints to ext_position control. That said, in this version of the sdk it should still be possible to change each joint individually to ext_position control using the 'set_operating_modes' ROS service.

> As this the function which publishes to joint_states. I checked joint_states topic and its values are between -2pi to 2pi. But the warning says that moveit expects it to be different. That's what I am not able to understand.

Technically, values should be from -pi to pi since this accounts for one full revolution (joint centered at 0 degrees with the ability to go to +/- 180 degrees). Also, the warnings you are seeing (about how the 'shoulder motor is not at the correct initial state' and 'Expected state: XXX, Actual State: XXX' ) is outputted by the interbotix_sdk node - not MoveIt. Essentially, when the interbotix_sdk node receives a joint trajectory command from MoveIt, it compares the joint values in the first waypoint to the current joint states. This is to make sure that they are the same or at least not too different (otherwise when the trajectory execution starts - the arm would jerk). However, in your case it seems that the joint commands in the first waypoint (and potentially the other waypoints) of the joint trajectory coming from MoveIt is roughly 6.7 times the current joint states. I'm not sure how this could be happening without seeing all the changes you made to the sdk. As an FYI, the line [here](https://github.com/Interbotix/interbotix_ros_arms/blob/master/interbotix_sdk/src/arm_obj.cpp#L1459) is where the current joint states are saved to a private variable called 'joint_states'. This is the variable that is used when comparing against the joint trajectory command being sent by MoveIt.

> I think this is where the problem lies. There maybe somewhere else from where moveit gets the joint states and I am missing that

MoveIt should only be getting the joints states from the /<robot_name>/joint_states topic, not anywhere else. If the arm showing up in Rviz seems to match the physical arm layout, then you should be good. Out of curiosity, if you just run the arm_run.launch file with Rviz, and manually manipulate the arm (torquing off the motors first), does the virtual model in Rviz mimic the real arm?

swinterbotix, 2020-07-02:
Closing due to lack of activity. Feel free to reopen if necessary.

closed 2020-07-02 by swinterbotix