Interbotix/interbotix_ros_arms #22: ImportError: No module named interbotix_moveit_interface.srv

jingfang, 2020-06-15:
When I ran the following command:
`roslaunch interbotix_moveit_interface moveit_interface.launch robot_name:=px150 use_gazebo:=true use_cpp_interface:=true`

I got an import error:
`
Traceback (most recent call last):
  File "/home/ros1/poc2020/ros1/src/interbotix_ros_arms/interbotix_examples/interbotix_moveit_interface/scripts/moveit_interface_gui", line 9, in <module>
    from interbotix_moveit_interface.srv import MoveItPlanRequest
ImportError: No module named interbotix_moveit_interface.srv
`

While I debug this error, I also noticed that "MoveItPlanRequest" doesn't exist in interbotix_moveit_interface/srv directory.

I'm willing to fix them and make a contribution if you could give me some instruction ( I'm new to this field).

swinterbotix, 2020-06-16:
Hi @jingfang,

Is it possible that you didn't catkin_make your repository? I just tried to run the command you suggested using the latest code in the repository and it worked three times in a row. I'd also recommend taking a look at the first section of [this](http://wiki.ros.org/rospy/Overview/Services) site that explains how MoveItPlanRequest and MoveItPlanResponse are autogenerated from MoveItPlan (probably at compile time).

If you did catkin_make and are still having issues, can you let me know what distro you're using? I tested this on ROS Kinetic/Ubuntu 16.04, but I can also test this on ROS Melodic/Ubuntu 18.04 if that's what you're using.

jingfang, 2020-06-16:
Hi @swinterbotix,

Thank you so much for your quick response and support!
Indeed, interbotix_moveit_interface wasn't build correctly.

My environment is Melodic/Ubuntu 18.04.
I'm using `catkin build` command at `//ros1/` directory, where interbotix_moveit_interface is located in
`//ros1/src/interbotix_ros_arms/interbotix_examples/interbotix_moveit_interface/`.

And the error I got is the following:
`Errors     << interbotix_moveit_interface:make /home/ros1/poc2020/ros1/logs/interbotix_moveit_interface/build.make.001.log
make[2]: *** No rule to make target '/opt/ros/melodic/lib/libmoveit_occupancy_map_monitor.so', needed by '/home/ros1/poc2020/ros1/devel/.private/interbotix_moveit_interface/lib/interbotix_moveit_interface/moveit_interface_node'.  Stop.
make[1]: *** [CMakeFiles/moveit_interface_node.dir/all] Error 2
make: *** [all] Error 2
cd /home/ros1/poc2020/ros1/build/interbotix_moveit_interface; catkin build --get-env interbotix_moveit_interface | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << interbotix_moveit_interface:make           [ Exited with code 2 ]`

P.S. Thank you for the link!

swinterbotix, 2020-06-16:
Did you make sure to install all the dependencies using the rosdep command (Step 4 in the Quickstart)?

swinterbotix, 2020-07-02:
Closing due to lack of activity. Feel free to reopen if necessary.

closed 2020-07-02 by swinterbotix