Interbotix/interbotix_ros_arms #23: interbotix_ws is really necessary?

Ktysai, 2020-07-09:
Hi there,

I want to use move it in order t control an PX100 arm. In order to reduce the number of work-spaces involved, can interbotix_ros_arms be put as an package in ws_moveit?

swinterbotix, 2020-07-09:
I assume ws_moveit is a local workspace on your computer? If yes, feel free to stick the repository in that workspace. Just make sure to run the rosdep command (Step 4) at the root of that workspace, and to change the workspace name appropriately when going through the Quick start.

Ktysai, 2020-07-09:
Thank you for the fast reply, the PC was overhelm by moveit build and I cannot refresh the page:)

I'll tell you soon the results!

I want to put a Kinect V2 in order to check the collisions. Have you tested such setup?

Cheers!

swinterbotix, 2020-07-09:
Glad to help! Regarding using a Kinect V2 to check for collisions, are you talking about doing that through MoveIt? There's a tutorial that walks through that at http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/perception_pipeline/perception_pipeline_tutorial.html. However, this is not implemented in the Interbotix ROS packages by default as the kits do not come with any depth cameras. Perhaps at a later point, I will experience with that. That said, I do have experience getting Intel RealSense Depth cameras up and running with ROS, just not with MoveIt.

Ktysai, 2020-07-09:
Yes, Kinect V2 & Moveit with Interbotix ROS packages is what I plan to use. I'll post here the results.

At first try I couldn't make the kinect send the proper info, so after a lot of broken packages, I've stated a new life: Ubuntu Bionic & ROS Melodic.

Intel RealSense Depth cameras, are pretty cool, for sure I'll invest in one. Especially that LiDAR Camera L515 :)) First to lay wth the PX100 arm & kinect!

It's ok to ask some question here regarding Interbotix arm & RGB-D camera or shall I start a new issues and to close this one

swinterbotix, 2020-07-09:
Yeah, the LiDAR L515 looks pretty dope! And even better, there's ROS support for it already despite it only being on preorder at Intel's website.

You should start a new issue for your camera related questions so that other people with similar questions can easily find it.

Ktysai, 2020-07-09:
Can we rename the thread " 'interbotix_ros_arms' used as simple package in MOVEIT/ROS Melodic" if you think it describes better the situation and to close the issue?

Thanks again,  **@swinterbotix** +1

swinterbotix, 2020-07-09:
You don't have to rename this thread. I think the current name is just fine.

closed 2020-07-09 by Ktysai