Interbotix/interbotix_ros_arms #24: Moveit & Gripper of PX100

Ktysai, 2020-07-13:
Hello!

On Moveit, after doing a setup with the Assistant,  the gripper even if I set it made of joints or links only one finger is moving.

As description files for the arm I use directly the XACRO file of px100. This is happening now in Melodic, but also with kinetic I was having this issue.

There is something I've overlooked?

All the best!

swinterbotix, 2020-07-13:
Is there a specific reason you're making a MoveIt package for the PX100 arm? By default, the PX100 arm is already compatible with MoveIt using the interbotix_moveit package.

Assuming you're creating your own MoveIt package from the px100.urd.xacro file, can you show me the generated SRDF?

Ktysai, 2020-07-14:
I use interbotix_moveit to learn from it. Main reason is to see what Moveit use in normal configuration. there's a lot to learn and I try to make it 'hands-on' as much as I can.

And I have to hide the arm from kids because the love to twist its joint.

Since you've wrote your comment I've looked at the SRDF file in the interbotix_moveit package and I've modified my SRDF in a text editor. Now It can be seen correctly in the MOVEIT setup assistant, but I have not save the preedited format

Here's one of the older version of the SRDF I have tried. It can be seen that only one finger can used. This happens in Assistant. I can edit manually and everything is fine, but for sure I overlook something.

`<?xml version="1.0" ?>

<robot name="px100" xmlns:xacro="http://www.ros.org/wiki/xacro">
    
    
    
    
    
    <xacro:arg name="robot_name" default=""/>
    <xacro:arg name="external_srdf_loc" default=""/>

    <xacro:property name="srdf_loc" value="$(arg external_srdf_loc)"/>

    <group name="interbotix_arm">
        <joint name="waist" />
        <joint name="shoulder" />
        <joint name="elbow" />
        <joint name="wrist_angle" />
        <joint name="ee_arm" />
    </group>
    <group name="interbotix_gripper">
        <link name="$(arg robot_name)/gripper_bar_link" />
        <link name="$(arg robot_name)/fingers_link" />
        <link name="$(arg robot_name)/ee_gripper_link" />
        <link name="$(arg robot_name)/left_finger_link" />
        <link name="$(arg robot_name)/right_finger_link" />
    </group>
    
    <group_state name="Home" group="interbotix_arm">
        <joint name="elbow" value="0" />
        <joint name="shoulder" value="0" />
        <joint name="waist" value="0" />
        <joint name="wrist_angle" value="0" />
    </group_state>
    <group_state name="Upright" group="interbotix_arm">
        <joint name="elbow" value="1.5708" />
        <joint name="shoulder" value="0" />
        <joint name="waist" value="0" />
        <joint name="wrist_angle" value="0" />
    </group_state>
    <group_state name="Sleep" group="interbotix_arm">
        <joint name="elbow" value="-1.5" />
        <joint name="shoulder" value="-1.88" />
        <joint name="waist" value="0" />
        <joint name="wrist_angle" value="-0.8" />
    </group_state>
    <group_state name="Open" group="interbotix_gripper">
        <joint name="left_finger" value="0.037" />
        <joint name="right_finger" value="-0.037" />
    </group_state>
    <group_state name="Closed" group="interbotix_gripper">
        <joint name="left_finger" value="0.015" />
        <joint name="right_finger" value="-0.015" />
    </group_state>
    <group_state name="Home" group="interbotix_gripper">
        <joint name="left_finger" value="0.0195" />
        <joint name="right_finger" value="-0.0195" />
    </group_state>
    
    <end_effector name="interbotix_gripper" parent_link="$(arg robot_name)/ee_arm_link" group="interbotix_gripper" />
    
    <disable_collisions link1="$(arg robot_name)/base_link" link2="$(arg robot_name)/shoulder_link" reason="Adjacent" />
    <disable_collisions link1="$(arg robot_name)/forearm_link" link2="$(arg robot_name)/gripper_bar_link" reason="Never" />
    <disable_collisions link1="$(arg robot_name)/forearm_link" link2="$(arg robot_name)/gripper_link" reason="Adjacent" />
    <disable_collisions link1="$(arg robot_name)/forearm_link" link2="$(arg robot_name)/gripper_prop_link" reason="Never" />
    <disable_collisions link1="$(arg robot_name)/forearm_link" link2="$(arg robot_name)/left_finger_link" reason="Never" />
    <disable_collisions link1="$(arg robot_name)/forearm_link" link2="$(arg robot_name)/right_finger_link" reason="Never" />
    <disable_collisions link1="$(arg robot_name)/forearm_link" link2="$(arg robot_name)/upper_arm_link" reason="Adjacent" />
    <disable_collisions link1="$(arg robot_name)/gripper_bar_link" link2="$(arg robot_name)/gripper_link" reason="Adjacent" />
    <disable_collisions link1="$(arg robot_name)/gripper_bar_link" link2="$(arg robot_name)/gripper_prop_link" reason="Adjacent" />
    <disable_collisions link1="$(arg robot_name)/gripper_bar_link" link2="$(arg robot_name)/left_finger_link" reason="Adjacent" />
    <disable_collisions link1="$(arg robot_name)/gripper_bar_link" link2="$(arg robot_name)/right_finger_link" reason="Adjacent" />
    <disable_collisions link1="$(arg robot_name)/gripper_link" link2="$(arg robot_name)/gripper_prop_link" reason="Adjacent" />
    <disable_collisions link1="$(arg robot_name)/gripper_link" link2="$(arg robot_name)/left_finger_link" reason="Never" />
    <disable_collisions link1="$(arg robot_name)/gripper_link" link2="$(arg robot_name)/right_finger_link" reason="Never" />
    <disable_collisions link1="$(arg robot_name)/gripper_prop_link" link2="$(arg robot_name)/left_finger_link" reason="Never" />
    <disable_collisions link1="$(arg robot_name)/gripper_prop_link" link2="$(arg robot_name)/right_finger_link" reason="Never" />
    <disable_collisions link1="$(arg robot_name)/left_finger_link" link2="$(arg robot_name)/right_finger_link" reason="Adjacent" />
    <disable_collisions link1="$(arg robot_name)/shoulder_link" link2="$(arg robot_name)/upper_arm_link" reason="Adjacent" />

    <xacro:if value="${srdf_loc != ''}">
      <xacro:include filename="${srdf_loc}"/>
    </xacro:if>

</robot>
`

swinterbotix, 2020-07-14:
@Ktysai,

I did a bit of digging and it looks like there might be a bug in the MoveIt Setup Assistant Wizard - most likely it's not geared to handle mimic joints (which I use for the 'right_finger' joint in the URDF) properly. Below is the raw version of the SRDF created by the Setup Assistant Wizard. If you pay attention to the 'interbotix_gripper' Named Group section, you'll see that MoveIt does not assign the negative version of the 'left_finger' joint for the 'right_finger' joint. Instead, it just sets it to 0. This is probably why you were not seeing the right_finger joint move at all. The workaround that I did was just to manually edit the 'right_finger' joint value in the SRDF to the negative of the 'left_finger' joint.

```
<?xml version="1.0" ?>
<!--This does not replace URDF, and is not an extension of URDF.
    This is a format for representing semantic information about the robot structure.
    A URDF file must exist for this robot as well, where the joints and the links that are referenced are defined
-->
<robot name="px100">
    <!--GROUPS: Representation of a set of joints and links. This can be useful for specifying DOF to plan for, defining arms, end effectors, etc-->
    <!--LINKS: When a link is specified, the parent joint of that link (if it exists) is automatically included-->
    <!--JOINTS: When a joint is specified, the child link of that joint (which will always exist) is automatically included-->
    <!--CHAINS: When a chain is specified, all the links along the chain (including endpoints) are included in the group. Additionally, all the joints that are parents to included links are also included. This means that joints along the chain and the parent joint of the base link are included in the group-->
    <!--SUBGROUPS: Groups can also be formed by referencing to already defined group names-->
    <group name="interbotix_arm">
        <joint name="waist" />
        <joint name="shoulder" />
        <joint name="elbow" />
        <joint name="wrist_angle" />
        <joint name="ee_arm" />
    </group>
    <group name="interbotix_gripper">
        <link name="px100/gripper_bar_link" />
        <link name="px100/fingers_link" />
        <link name="px100/ee_gripper_link" />
        <link name="px100/left_finger_link" />
        <link name="px100/right_finger_link" />
    </group>
    <!--GROUP STATES: Purpose: Define a named state for a particular group, in terms of joint values. This is useful to define states like 'folded arms'-->
    <group_state name="Home" group="interbotix_arm">
        <joint name="elbow" value="0" />
        <joint name="shoulder" value="0" />
        <joint name="waist" value="0" />
        <joint name="wrist_angle" value="0" />
    </group_state>
    <group_state name="Upright" group="interbotix_arm">
        <joint name="elbow" value="1.5708" />
        <joint name="shoulder" value="0" />
        <joint name="waist" value="0" />
        <joint name="wrist_angle" value="0" />
    </group_state>
    <group_state name="Sleep" group="interbotix_arm">
        <joint name="elbow" value="-1.5" />
        <joint name="shoulder" value="-1.88" />
        <joint name="waist" value="0" />
        <joint name="wrist_angle" value="-0.8" />
    </group_state>
    <group_state name="Open" group="interbotix_gripper">
        <joint name="left_finger" value="0.037" />
        <joint name="right_finger" value="0" />
    </group_state>
    <group_state name="Closed" group="interbotix_gripper">
        <joint name="left_finger" value="0.015" />
        <joint name="right_finger" value="0" />
    </group_state>
    <group_state name="Home" group="interbotix_gripper">
        <joint name="left_finger" value="0.0195" />
        <joint name="right_finger" value="0" />
    </group_state>
    <!--END EFFECTOR: Purpose: Represent information about an end effector.-->
    <end_effector name="interbotix_gripper" parent_link="px100/ee_arm_link" group="interbotix_gripper" parent_group="interbotix_arm" />
    <!--DISABLE COLLISIONS: By default it is assumed that any link of the robot could potentially come into collision with any other link in the robot. This tag disables collision checking between a specified pair of links. -->
    <disable_collisions link1="px100/base_link" link2="px100/shoulder_link" reason="Adjacent" />
    <disable_collisions link1="px100/forearm_link" link2="px100/gripper_bar_link" reason="Never" />
    <disable_collisions link1="px100/forearm_link" link2="px100/gripper_link" reason="Adjacent" />
    <disable_collisions link1="px100/forearm_link" link2="px100/gripper_prop_link" reason="Never" />
    <disable_collisions link1="px100/forearm_link" link2="px100/left_finger_link" reason="Never" />
    <disable_collisions link1="px100/forearm_link" link2="px100/right_finger_link" reason="Never" />
    <disable_collisions link1="px100/forearm_link" link2="px100/upper_arm_link" reason="Adjacent" />
    <disable_collisions link1="px100/gripper_bar_link" link2="px100/gripper_link" reason="Adjacent" />
    <disable_collisions link1="px100/gripper_bar_link" link2="px100/gripper_prop_link" reason="Adjacent" />
    <disable_collisions link1="px100/gripper_bar_link" link2="px100/left_finger_link" reason="Adjacent" />
    <disable_collisions link1="px100/gripper_bar_link" link2="px100/right_finger_link" reason="Adjacent" />
    <disable_collisions link1="px100/gripper_link" link2="px100/gripper_prop_link" reason="Adjacent" />
[... cut ...]
    <disable_collisions link1="px100/gripper_prop_link" link2="px100/left_finger_link" reason="Never" />
    <disable_collisions link1="px100/gripper_prop_link" link2="px100/right_finger_link" reason="Never" />
    <disable_collisions link1="px100/gripper_prop_link" link2="px100/upper_arm_link" reason="Never" />
    <disable_collisions link1="px100/left_finger_link" link2="px100/right_finger_link" reason="Adjacent" />
    <disable_collisions link1="px100/shoulder_link" link2="px100/upper_arm_link" reason="Adjacent" />
</robot>
```

Ktysai, 2020-07-15:
It's perfect! The assistant makes mistakes sometimes even on setting joint limits, but it's still a good point to start.

Sorry for the delayed response, the kinect install on Melodic and its calibration has eaten up my hobby time.

closed 2020-07-15 by Ktysai

swinterbotix, 2020-07-15:
Np! Didn't know it also made mistakes with the joint limits file. The one it generated for me though seemed to be correct.