Interbotix/interbotix_ros_arms #25: Add a new joystick/gamepad to an interbotix arm

Ktysai, 2020-07-17:
Hi there!

I've tried to connect an[ Thrustmater T-wireless](http://www.thrustmaster.com/products/t-wireless-3-1-rumble-force) gamepad that it's compatible with PS3 & PS2. I've tried in MOveit, and I've realized
that some file had to be modified. After that, I've seen that in MOveit the joystick is used just for goals, not realtime movement...

So I've remember that in the interbotix github was a joystick control that never worked for my gamepad. 
And I was in love when I've seen the package **interbotix_joy_control** ... again :)))

I've modified that cool file :+1: **joy_control_node.cpp** by adding a new class **thrustM** and setting it up.
In order to make it work, I've taken a part of code from PS3 & Xbox.
After some setting up the launch file with proper argument finally the robot was controlled in real-time. A very good point for Interbotix :)

**I have some issues though:**
1. Even if I change the designation of a command to another button, every time the command is issued from the first button.

Example, {"GRIPPER_PWM_DEC", 8}, so button '8' has to command the speed of the gripper, but button '0' does that.

2. The mappings aren't quite correct, example {"WAIST_CCW", 4}, in reality the command is set by button '6'.

3. It would be cool to add speed control of the robot joints. But all the mappings tested haven't been able to do that.

The code is here:
`
#include <ros/ros.h>
#include <sensor_msgs/Joy.h>
#include "interbotix_joy_control/ArmJoyControl.h"

// PS3 Controller button mappings
static const std::map<std::string, int> ps3 = {{"GRIPPER_PWM_DEC", 0}, // buttons start here
                                               {"GRIPPER_OPEN", 1},
                                               {"GRIPPER_PWM_INC", 2},
                                               {"GRIPPER_CLOSE", 3},
                                               {"EE_Y_INC", 4},
                                               {"EE_Y_DEC", 5},
                                               {"WAIST_CCW", 6},
                                               {"WAIST_CW", 7},
                                               {"SLEEP_POSE", 8},
                                               {"HOME_POSE", 9},
                                               {"FLIP_EE_X", 11},
                                               {"FLIP_EE_ROLL", 12},
                                               {"SPEED_INC", 13},
                                               {"SPEED_DEC", 14},
                                               {"SPEED_COURSE", 15},
                                               {"SPEED_FINE", 16},
                                               {"EE_X", 0},            // axes start here
                                               {"EE_Z", 1},
                                               {"EE_ROLL", 3},
                                               {"EE_PITCH", 4}};

// PS4 Controller button mappings
static const std::map<std::string, int> ps4 = {{"GRIPPER_PWM_DEC", 0}, // buttons start here
                                               {"GRIPPER_OPEN", 1},
                                               {"GRIPPER_PWM_INC", 2},
                                               {"GRIPPER_CLOSE", 3},
                                               {"EE_Y_INC", 4},
                                               {"EE_Y_DEC", 5},
                                               {"WAIST_CCW", 6},
                                               {"WAIST_CW", 7},
                                               {"SLEEP_POSE", 8},
                                               {"HOME_POSE", 9},
                                               {"FLIP_EE_X", 11},
                                               {"FLIP_EE_ROLL", 12},
                                               {"EE_X", 0},            // axes start here
                                               {"EE_Z", 1},
                                               {"EE_ROLL", 3},
                                               {"EE_PITCH", 4},
                                               {"SPEED_TYPE", 6},
                                               {"SPEED", 7}};

// Xbox 360 Controller button mappings
[... 184 lines cut ...]
      pub_joy_cmd.publish(joy_cmd);
  prev_joy_cmd = joy_cmd;
}

int main(int argc, char **argv)
{
  ros::init(argc, argv, "interbotix_joy_control_node");
  ros::NodeHandle n;
  ros::param::get("~threshold", threshold);
  ros::param::get("~controller", controller_type);
  if (controller_type == "xbox360")
    cntlr = xbox360;
  else if (controller_type == "ps3")
    cntlr = ps3;
  else if (controller_type == "thrustM") //!.!.o mod
    cntlr = thrustM;
  else
    cntlr = ps4;

  sub_joy_raw = n.subscribe("joy", 100, joy_state_cb);
  pub_joy_cmd = n.advertise<interbotix_joy_control::ArmJoyControl>("joy/commands", 100);
  ros::spin();
  return 0;
}
`

swinterbotix, 2020-07-20:
Hi @Ktysai,

Nice to see someone on their way to incorporating another joystick! About your issue - I've never used a thrust master controller before, but your first step would be to check that the [joy](http://wiki.ros.org/joy) node works on Linux with that type of controller. Essentially, do this (after paring the joystick with the Linux computer)

1. Open a terminal and type `roscore`
2. Open a second terminal and type `rosrun joy joy_node`
3. Open a third terminal and type `rostopic echo /joy`

Now, run through all the buttons and sticks on your thrust master controller one by one, and pay attention to the index by which they appear in the 'buttons' and 'axes' arrays. These are the mappings you want to use in the joy_node.cpp file.
Also note that you will have to edit some of the 'if' - 'else' conditions in the file to include your controller. For example, when checking the 'speed_cmd', you will have to include 'thrustM' in the condition as well, as by default, it's excluded.

Ktysai, 2020-07-21:
OK, @**swinterbotix**!

I'll recheck the mappings in the joy rostopic, and with `jstest -gtk`. If only **joy_node.cpp file** has the control over all the mappings makes things easier. 

The joystick works fine for controlling a differential two motor wheels robot, like turtlebot.

Ktysai, 2020-07-21:
I've checked the joy topic and indeed the buttons are observed correctly. Tried diffident mappings... but nada

Still have the issues regarding the mappings of the buttons, even if button '0' has other designation, all the time it will decrease the GRIPER PWM. Is like there is a master mapping of the joystick, that has priority.

Anyway, I'll leave it like that because I can control the arm.

The default joint speed, is a little bit too high for my liking, so I've set the value '0.75', for the 'current' variable in `joy_robot_control.py`. Is not nice, but it works for now.

Btw, the code for the control of the robot is open source? Cannot be a part of Moveit? if I understand correctly, at the moment Moveit lacks this control option.

swinterbotix, 2020-07-21:
Strange that you're not able to map the gripper pwm inc/dec or the joint speed buttons correctly. Without actually having a ThrustM controller, I can't really help you debug that unfortunately.

Yes, the code for all of this is open source. If someone wanted to, they could create a pipeline to do this via MoveIt. I'd image the structure would be the joy_node (responsible for publishing joystick values to a topic) -> joy_mapper_node (responsible for mapping joystick buttons to desired Cartesian motion) -> robot_moveit_control_node (responsible for planning motion and executing using MoveIt's Move Group API).

However, in my experience, MoveIt is terrible at realtime motion and more often than not - will yell at you saying it can't find a desired pose - essentially making the whole joystick control node useless. You might be able to use this joystick pipeline to define desired waypoints for an arm to follow, and in the unlikely miracle that MoveIt actually is able to plan all the waypoints successfully - execute them - but it wouldn't be in real time.

That's one of the reasons I created my own Cartesian joystick node. It can be used in realtime and will never fail in finding the next desired goal pose (unless you're moving the end-effector out of the physical workspace of the arm).

Anyway, that's my rant on MoveIt. Thanks for listening :)

Ktysai, 2020-07-22:
No problem, it was a useful rant :1st_place_medal: 
Just after progressively I understand more about ROS, and in this particular case about Moveit, I can really appreciate add-ons for it.

Your idea is good, and also the implementation.

I'll use the node as it is, and maybe take a look at those mappings.

To keep the discussion into the joystick area, what model do you advise me to invest?
I've tried with ones borrowed from friends: 
- Joystick for Xbox one - very smooth, precise. It was hard until I get the proper drivers, and some mappings;
- Joystick PS3 - less precise than the one for Xbox one, works out of the box;
- Joystick Xbox 360,  works almost out of the box, the precision was weird... it was seriously warn out;
- Joystick Thrustmater T-wireless, works mostly out of the box, and have the best coverage range from all tested. Lacks precision, but resists to the (un)wanted attacks of my twins :) 

I haven't tried PS4 yet :)

If you have the time, can you please try mapping the joystick you have?

Now can I use your Cartesian joystick node to:
- to put the robot in a desired position and get the joint position value?
- can it be run almost 'simultaneous' with Moveit? To run the Setup Assistant for example, and to set several position that are wanted?

swinterbotix, 2020-07-22:
I've tested my code with both the PS3 and PS4 controllers, and verified that they work correctly. A different user added the Xbox360 controller, but I have not actually tested it. In my opinion, a PS4 controller is the way to go just because it's easily available, and I know it works (at least the official SONY version). It's hard to find an official SONY PS3 controller nowadays - but you can also go that route if money is an issue.

swinterbotix, 2020-08-05:
Closing due to lack of activity. Feel free to reopen if you still have questions.

closed 2020-08-05 by swinterbotix