Interbotix/interbotix_ros_arms #26: Dealing with max effort reached

TaekedeHaan, 2020-08-12:
Sometimes, when an object is grasped incorrectly the action is stopped due to the maximum effort:
```
Node: /px150/arm_node
Time: 15:40:53.121059239 (2020-08-12)
Severity: Info
Published Topics: /px150/arm_controller/follow_joint_trajectory/feedback, /px150/arm_controller/follow_joint_trajectory/result, /px150/arm_controller/follow_joint_trajectory/status, /px150/gripper_controller/follow_joint_trajectory/feedback, /px150/gripper_controller/follow_joint_trajectory/result, /px150/gripper_controller/follow_joint_trajectory/status, /px150/joint_states, /rosout

Gripper trajectory server preempted itself since max effort reached.

Location:
/home/taeke/catkin_ws/src/interbotix_ros_arms/interbotix_sdk/src/arm_obj.cpp:arm_gripper_trajectory_action_callback:1695
```

Often when this happens the Dynamixel on which the overload has occurred (generally the gripper) goes into an error mode with a red blinking LED. If I try to re-grasp I receive the following message:

```
Node: /px150/arm_node
Time: 15:59:09.046790848 (2020-08-12)
Severity: Error
Published Topics: /px150/arm_controller/follow_joint_trajectory/feedback, /px150/arm_controller/follow_joint_trajectory/result, /px150/arm_controller/follow_joint_trajectory/status, /px150/gripper_controller/follow_joint_trajectory/feedback, /px150/gripper_controller/follow_joint_trajectory/result, /px150/gripper_controller/follow_joint_trajectory/status, /px150/joint_states, /rosout

[DynamixelWorkbench] Failed to set goal position!

Location:
/home/taeke/catkin_ws/src/interbotix_ros_arms/interbotix_sdk/src/arm_obj.cpp:arm_check_error:1796
```
Currently the only way I can continue is by switching the robot off and on. However I would like to:

- Identify which Dynamixel is causing this error
- Clear the error such that I can continue

However, I do not know how to implement this. I was wondering if you know how this could be approached?

TaekedeHaan, 2020-08-12:
Apparently there is a service `get_motor_register_values` available to read the the registers, which will enable me to detect the Overload Error.

I can see that the [dynamixel workbench](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#api-references) also provides `reboot()` function, which what I need. Would it be interesting to add this to the other services provided by arm_obj.cpp?

swinterbotix, 2020-08-12:
Hi @TaekedeHaan,

Sounds like you're using MoveIt's API to open and close the gripper. Specifically, you're probably using the 'Open' and "Closed' Named Poses, and you're seeing the gripper motor go into an Overload state a few seconds after commanding the gripper to close (LED on the gripper servo starts blinking). The reason for that is because the 'Closed' pose is defined as when the gripper fingers are essentially touching each other. However, you're probably picking up a rigid block. So what happens is the gripper motor is successful at gripping the block, but the motor still tries to close the fingers since it's trying to get to the predefined 'Closed' position. Unfortunately, it can't since the block is in the way. So the gripper servo is a bit too smart - it sees that no matter how much it tries to turn the motor, it won't budge. After a few seconds of this, it thinks the motor is stalled, and torques off (and goes into an Overload state) as a safety precaution.

The way my code 'should' get around this issue is by detecting a spike in current (which I define as 600 mA), and commanding the motor to hold its current position instead of trying to go to the fully 'Closed' position. Obviously, that limit was a bit high and should probably be lowered to prevent the motor from Overloading. Feel free to lower it to 400 mA over [here](https://github.com/Interbotix/interbotix_ros_arms/blob/master/interbotix_sdk/include/interbotix_sdk/arm_obj.h#L37), and recompile by running 'catkin_make' at the root of the workspace.

Another approach would be to command a custom gripper position as opposed the the predefined 'Closed' position. To give some reference, fully closed is 0.015 meters (from the middle of a gripper finger to the center of the servo horn). Fully open is 0.037 meters. So commanding a position that closes just enough to pick up your object (perhaps 0.025?), but not close any further should make this issue go away.

Yet another approach would be to do away with using MoveIt to open/close the gripper at all (as is discussed [here](https://answers.ros.org/question/198706/how-to-plan-moves-with-gripper-attached-to-arm/) in Approach 1). Instead, I'd recommend putting the gripper servo in PWM mode at startup and command PWM signals to the `/<robot_name>/single_joint/command` topic (+/- 350 should do the trick!) to open and close the gripper (you could also send to the `/robot_name>/gripper/command` topic, but I'm planning on getting rid of this topic in a future update). This way, you will be able to pick up any sized object firmly without worrying about the motor trying to get to a specific position - since there _is_ no specific position for the motor to go to! It's just following a PWM command. Note though that you will need to check the gripper position in your own code repeatedly and send PWM commands of 0 to prevent the motor from turning past the fully open or fully closed position. That's what I do in my joystick control code [here](https://github.com/Interbotix/interbotix_ros_arms/blob/master/interbotix_examples/interbotix_joy_control/scripts/joy_robot_control.py#L213) if you want to reference it.

About adding a 'reboot' service - that actually sounds like a good idea. I'll consider adding that in a future update.

On a different note, can you tell me how you got those debugging messages to appear? It would definitely save me time debugging future issues if I knew which line was outputting an error message lol.

TaekedeHaan, 2020-08-13:
Thanks a lot for you elaborate answer! I indeed use MoveIt for commanding the gripper. Two knobs were attached to the gripper which should enclose the object I am trying to grasp. Sometimes the gripper is placed incorrectly, meaning that the knobs get stuck on the object, and the overload error is triggered.

I will try to lower the maximum effort, this seems like a good starting point. Also PWM control seems like a great idea, and much more suitable for my situation anyway.

I created the messages with [rqt_console](http://wiki.ros.org/rqt_console). You can open the message viewer by double clicking a message. It is very useful for debugging!

TaekedeHaan, 2020-08-13:
I implemented a reboot service, which is based on the already present `arm_get_firmware_register_values()` method https://github.com/Interbotix/interbotix_ros_arms/compare/master...TaekedeHaan:reboot_service

I can use it to get out of Shutdown state. After a reboot all motors are reinitialized using `arm_init_operating_modes()`. A cleaner implementation might be be to set the operating modes of the rebooted motors to their modes before the reboot. However, I think this would require adding an `arm_set_operating_modes()` method.

swinterbotix, 2020-08-13:
> I created the messages with rqt_console. You can open the message viewer by double clicking a message. It is very useful for debugging!

Awesome!

Yeah, it would be a good idea to remember the operating mode in order to set it later.

TaekedeHaan, 2020-08-27:
As discussed [here](https://github.com/ROBOTIS-GIT/dynamixel-workbench/issues/235) the Hardware Error Status (70) should be read to obtain the shutdown reason. However when I try to do this whenever an error occurred, the provided `get_motor_register_values` service gives an error e.g.:
```
`Failed to read value on items[Hardware_Error_Status] from Dynamixel[ID : 2]`
```
This means that `dxl_wb.itemRead` returns false, and thus was unsuccessful as can be seen on [this](https://github.com/Interbotix/interbotix_ros_arms/blob/59517af0b776033de0989d26bd3103f50b267cb1/interbotix_sdk/src/arm_obj.cpp#L1302) line. Do you have any idea why this is happening? The strange thing is that with the Dynamixel Wizard I am able to read the  Hardware_Error_Status values. I could also try to ask it on the dynamixel-workbench github page, if you think the problem lies there.

swinterbotix, 2020-08-27:
I've been playing around with that service. It seems that I can only successfully read that register if no error is present. However, once an Overload error occurs, the itemRead function doesn't work for any register. Every time I try, the Dynamixel Workbench outputs an error message saying...

`[RxPacketError] Hardware error occurred. Check the error at Control Table (Hardware Error Status)!`

Seems like a catch 22. You can only read the register if the motor is not in an error state. But if it is in an error state, then you can't read anything?

This sounds like an issue for the Dynamixel Workbench folks!

swinterbotix, 2020-08-27:
As an FYI, I just finished making a reboot service for the next update. It allows you to reboot a single servo or a specified group of servos, as well as specify if you'd like the servos to be torqued on after reboot. Regarding setting operating modes... I totally forgot about this, but all EEPROM registers in the servos retain their values during power cycles. As a result, the motor(s) will automatically be set to the prior operating mode, drive mode, etc... as before the reboot. However, registers like 'Profile_Velocity', 'Profile_Acceleration', and PID gains will be reset to their factory defaults.

swinterbotix, 2020-09-22:
Closing this issue, as the itemRead function error is a problem with the Dynamixel Workbench API, not this one. Also, the new Reboot ROS service has already been implemented in the upcoming update. Feel free to reopen for any other questions!

closed 2020-09-22 by swinterbotix

febert, 2021-01-18:
Where can I find this reboot service? It doesn't seem to be in the melodic branch on either this repo or the newer repo.

swiz23, 2021-01-18:
The reboot service is only available in the newer repo. You can find information about it at https://github.com/Interbotix/interbotix_ros_core/tree/main/interbotix_ros_xseries/interbotix_xs_sdk#services.

The Dynamixel based arms in the interbotix_ros_mainpulators repo build ontop of the interbotix_ros_core repo.