Interbotix/interbotix_ros_arms #27: Error when runnning arm_node on Ubuntu 18.04 with ROS Melodic

hristo-vrigazov, 2020-08-12:
Hi, I am using ROS Melodic, but I get an error when the arm_node is getting run. It worked without problems on the Raspberry Pi + PS4 controller. So it is quite possible that I am doing something wrong. Basically I followed the instructions and I get an error on Step 10. Here's some logs:

```
λ hvrigazov [~/robo/interbotix_ws] → roslaunch interbotix_sdk arm_run.launch robot_name:=px150 robot_model:=px150
... logging to /home/hvrigazov/.ros/log/abe04a4a-dcc0-11ea-a3c1-309c23209558/roslaunch-hvrigazov-MS-7A91-25533.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hvrigazov-MS-7A91:37843/

SUMMARY
========

PARAMETERS
 * /px150/arm_node/arm_operating_mode: position
 * /px150/arm_node/arm_profile_acceleration: 15
 * /px150/arm_node/arm_profile_velocity: 131
 * /px150/arm_node/gripper_operating_mode: position
 * /px150/arm_node/gripper_profile_acceleration: 0
 * /px150/arm_node/gripper_profile_velocity: 0
 * /px150/arm_node/motor_configs: /home/hvrigazov/r...
 * /px150/arm_node/port: /dev/ttyDXL
 * /px150/arm_node/robot_model: px150
 * /px150/arm_node/robot_name: px150
 * /px150/arm_node/use_default_gripper_bar: True
 * /px150/arm_node/use_default_gripper_fingers: True
 * /px150/arm_node/use_moveit: False
 * /px150/arm_node/use_pid_cntlrs: False
 * /px150/arm_node/use_time_based_profile: False
 * /px150/robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.6

NODES
  /px150/
    arm_node (interbotix_sdk/arm_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [25547]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to abe04a4a-dcc0-11ea-a3c1-309c23209558
process[rosout-1]: started with pid [25558]
started core service [/rosout]
process[px150/robot_state_publisher-2]: started with pid [25565]
process[px150/rviz-3]: started with pid [25566]
process[px150/arm_node-4]: started with pid [25567]
Robot name: px150
/home/hvrigazov/robo/interbotix_ws/src/interbotix_ros_arms/interbotix_sdk/config/px150.yaml
[ INFO] [1597253067.374769085]: ID : 1, Model Number : 1060
[ INFO] [1597253067.410712184]: ID : 2, Model Number : 1060
[ INFO] [1597253067.445674002]: ID : 3, Model Number : 1060
[ INFO] [1597253067.480672046]: ID : 4, Model Number : 1060
[ INFO] [1597253067.515697707]: ID : 5, Model Number : 1060
[ INFO] [1597253067.550673990]: ID : 6, Model Number : 1060
[ INFO] [1597253067.585670004]: ID : 7, Model Number : 1060
[ERROR] [1597253067.653944120]: [TxRxResult] There is no status packet!
[ERROR] [1597253067.654003113]: Can't find Dynamixel ID '8'
[ INFO] [1597253067.708715720]: Robot torqued off!
[ERROR] [1597253068.170669249]: [DynamixelDriver] Failed to get the Tool
[ERROR] [1597253068.170736716]: Failed to write value[4095] on items[Max_Position_Limit] to [ID : 8]
[ INFO] [1597253068.170800096]: [DynamixelDriver] Succeeded to add sync write handler
[ INFO] [1597253068.170829907]: [DynamixelDriver] Succeeded to add sync write handler
[ INFO] [1597253068.170858142]: [DynamixelDriver] Succeeded to add sync write handler
[ INFO] [1597253068.560668013]: Arm joints set to position control with a profile velocity of 131 and a profile acceleration of 15.
[ERROR] [1597253068.560776967]: [DynamixelDriver] Failed to get the Tool
[ INFO] [1597253068.560832619]: gripper set to position control with a profile velocity of 0 and a profile acceleration of 0.
[ INFO] [1597253068.572364066]: All motors ready to go!
[ERROR] [1597253068.617491853]: [TxRxResult] There is no status packet!
[ERROR] [1597253068.617564652]: groupSyncRead getdata failed
[ERROR] [1597253068.617585973]: groupSyncRead getdata failed
[ERROR] [1597253068.617607590]: groupSyncRead getdata failed
[px150/arm_node-4] process has died [pid 25567, exit code -11, cmd /home/hvrigazov/robo/interbotix_ws/devel/lib/interbotix_sdk/arm_node __name:=arm_node __log:=/home/hvrigazov/.ros/log/abe04a4a-dcc0-11ea-a3c1-309c23209558/px150-arm_node-4.log].
log file: /home/hvrigazov/.ros/log/abe04a4a-dcc0-11ea-a3c1-309c23209558/px150-arm_node-4*.log

```

I don't understand this error. I googled and it seems to be related to the Dynamixel? Should I install something else apart from ROS Мelodic on the connected PC? I though of installing Dynamixel SDK, but based on the information here: http://support.interbotix.com/ it looks like its purpose is different.

I just received the physical arm today, have not installed anything apart from ROS. Everything runs correctly in the simulator and with the Raspberry Pi.

swinterbotix, 2020-08-12:
Hi @hristo-vrigazov,

From the error message, it seems that the arm_node can't find the 'gripper' servo (which is ID'd #8). Can you verify that the 3pin cable going from the gripper servo to the wrist_rotate servo is snugly connected?

Also, if the LED on the gripper servo is blinking, just doing a power cycle (unplugging and plugging power back in to the arm) and rerunning the node should work.

Did you say that this exact arm worked fine on the raspberry pi? Or was that a different arm? If it was a different arm, it's possible that the gripper servo was not ID'd correctly on the one you just received. Either way, I'd recommend installing the [Dynamixel Wizard 2.0](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#install-linux) tool. It's great for debugging purposes and works by itself without any ROS packages. Before scanning for motors, go to the Options tab and make sure that 1000000 bps is selected, the correct port is selected, and that IDs at least in the range between 1 and 8 are selected. If needed, it would be a simple matter of finding the gripper motor in the Wizard and changing its ID to 8 if that's the issue.

As an FYI, I don't believe there is any other software needed to get the ROS packages to work. From the fact that the node was able to find the first 7 servos with no issue tells me that all the needed software is installed. In case you're curious, a ros wrapper containing the dynamixel sdk (called ros-melodic-dynamixel-sdk) should've been automatically installed when you ran step #4 in the Quickstart.

hristo-vrigazov, 2020-08-12:
It was the same arm... and yes, just the cable of the wrist_rotate was off, I feel stupid :D Thanks!

closed 2020-08-12 by hristo-vrigazov

swinterbotix, 2020-08-12:
No prob man! Glad to help