Interbotix/interbotix_ros_arms #29: [TxRxResult] Incorrect status packet error

dchan-hatci, 2020-08-31:
Hi there - 

I just got my PX150 arm and I'm very excited to get it going! I'm having an issue getting ROS up and running. I'm going through the quickstart, and had no issues until the line 

'roslaunch interbotix_sdk arm_run.launch robot_name:=px150'

at which point I get continuous errors:
'[ERROR] [long number]" [TxRxResult] Incorrect status packet!
[ERROR] [long number]": groupSyncRead getdata failed'

I can see the /dev/ttyDXL on my /dev folder, and the torque on and off commands work, so I know the communication isn't totally broken. The Gazebo tracking does not work however so I'm thinking the motors are not correctly transmitting their packets. I'm running ROS Melodic on Ubuntu 18.04. I would appreciate any advice on how to debug this issue. Thank you!

swinterbotix, 2020-08-31:
Can you attach a snapshot of the roslaunch command followed by the errors in the terminal? Also, did you make sure to plug 12V power into the arm?

Also, Gazebo should not be running if you are controlling the physical arm. The idea of Gazebo is to simulate the robot so you can try things out before working on the physical robot.

dchan-hatci, 2020-08-31:
Thanks for the quick response –

12V power is plugged in, and I double checked all the three pin cables are snugly connected. I reran the command to start up the robot, screenshots below. It gets to the point where it says "All motors ready to go!".

Sorry if I was unclear re:Gazebo, I meant whatever the software that is supposed to track the robot (maybe RViz?). Let me know if I can help clarify the issue further. 

[image]
[image]

swinterbotix, 2020-08-31:
What is your computer platform?

dchan-hatci, 2020-08-31:
I'm running Ubuntu 18.04 in VirtualBox on my Windows 10 computer.

swinterbotix, 2020-08-31:
Hmmmm, Ok.

I've never tested our ROS packages in a VirtualBox before. I've heard that there can be issues communicating via USB ports. However, considering the fact that you are able to open the port and ping the servos - that implies you can do that to some degree. Perhaps there is some other plugin needed to get USB communication to be smoother?

swinterbotix, 2020-08-31:
According to the article [here](https://robocademy.com/2020/05/17/best-4-ways-to-install-ubuntu-for-ros/), it seems to say that you will have better luck working with USB connected devices using VMWare as opposed to Virtualbox.

Scroll down to the heading title _Which VM software will be good for working on R.O.S?_

dchan-hatci, 2020-08-31:
Ok thank you, I will try VMWare and do some digging on the usb communication. It's strange because the setup seems to be working  - maybe the high baud rate creates problems with the virtual machine (?)

I am getting a Nvidia Jetson Nano soon and will test in Linux on that environment to make sure it's not a hardware issue. If anyone else has suggestions, please let me know.

swinterbotix, 2020-08-31:
As a cheaper alternative, you could also test it out on a Raspberry Pi 3b+. Instructions to do so can be found at https://github.com/Interbotix/interbotix_ros_arms_pi#raspberry-pi-setup. I've tested our arms on that platform before and know it should work.

dchan-hatci, 2020-09-21:
To follow up on this issue, as @swinterbotix suggested, I tried using a different virtual machine (VMWare instead of VirtualBox) and it's working without issues on my PC. Thanks for the help!

closed 2020-09-21 by dchan-hatci