Interbotix/interbotix_ros_arms #31: Error Opening Serial Port

PeterJochem, 2020-09-18:
I had success getting my ReactorX up and running but then it stopped working. I seem to have an issue with ports. I though maybe someone could point me in the right direction? The error message is the same as the one from post at https://github.com/ROBOTIS-GIT/open_manipulator/issues/82 . 
[image]

swinterbotix, 2020-09-18:
Can you verify the following?

1. You've done Steps 7 and 8 in the Quick start, and you can see 'ttyDXL' show up in the /dev directory

2. The microUSB cable connecting the U2D2 to your laptop is not loose but firmly connected

3. No other application on your computer is using the port. For example, if you are running the Dynamixel Wizard 2 tool, and then try to run the ROS packages, you will get this error since it's already being used. You may also have a previous ROS session still using the port. To get rid of it, do `pkill ros` in the terminal.

4. If you have another udev rule in the /etc/udev/rules.d directory with a higher number than 10, it may be restricting your user's port permissions somehow. In that case, change the interbotix udev rule to 99-interbotix-udev.rules instead of 10-interbotix-udev.rules. Unplug/replug and try again.

PeterJochem, 2020-09-18:
Thanks! It was the fourth one. Looks good now

closed 2020-09-18 by PeterJochem