Interbotix/interbotix_ros_arms #32: RX150

sayyidabeegam, 2020-10-12:
When I try moveit_python_interface (https://github.com/Interbotix/interbotix_ros_arms/tree/master/interbotix_examples/interbotix_moveit_interface/scripts) I am getting error 

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
**Failed to import pyassimp, see https://github.com/ros-planning/moveit/issues/86 for more info
============ Press `Enter` to begin the tutorial by setting up the moveit_commander (press ctrl-d to exit) ...

[ERROR] [1602483496.283697682]: Robot model parameter not found! Did you remap 'robot_description'?
Traceback (most recent call last):
  File "moveit_python_interface", line 459, in <module>
    main()
  File "moveit_python_interface", line 409, in main
    tutorial = MoveGroupPythonIntefaceTutorial()
  File "moveit_python_interface", line 59, in __init__
    robot = moveit_commander.RobotCommander()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/moveit_commander/robot.py", line 151, in __init__
    self._r = _moveit_robot_interface.RobotInterface(robot_description, ns)
RuntimeError: RobotInterfacePython: invalid robot model**

//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

I remapped the robot model since it already gave the group name and all , I have given the robot model and DOF. But still showing the error. 
Also if I want some intermediate pose along with  "pose_goal", how can I provide it?

swinterbotix, 2020-10-12:
This can happen if you're improperly launching the *moveit_interface.launch* file. You need to set either the `use_actual`, `use_gazebo`, or `use_fake` launch file arguments to true. Otherwise, the launch file won't know if it's supposed to work with the actual robot or a simulated one. For example, to work on the physical robot, you would do...

`roslaunch interbotix_moveit_interface moveit_interface.launch robot_name:=rx150 use_actual:=true use_python_interface:=true`

Alternatively, if you're trying to run just the moveit_python_interface node using the `rosrun` command, I'd recommend against that - as you need to pass in some parameters and remap some topics. The snipped in the launch file that actually launches the node and which also shows the parameters that need to be passed in can be found [here](https://github.com/Interbotix/interbotix_ros_arms/blob/59517af0b776033de0989d26bd3103f50b267cb1/interbotix_examples/interbotix_moveit_interface/launch/moveit_interface.launch#L47).

About the pyassimp error...

This is a known bug that is due to Linux installing an outdated version of pyassimp (version 3.2). To fix it, install a more updated version using pip. Just open a terminal and type `pip install --user pyassimp`. If that's still not working for you, you can try the solution proposed at  https://github.com/ros-planning/moveit/issues/86#issuecomment-670519295.

closed 2020-10-13 by sayyidabeegam

sunnyshi0310, 2021-01-18:
Hi, I want to followup this question. So what we could do to send several commands to the robot using moveit without opening the RViz and set the goals manually. Could we write a python script and publish to some topics (I saw it's "display_planned_path" or something?) and achieve the goal directly?

Thanks for any useful information.

swiz23, 2021-01-18:
The answer to @sunnyshi0310's question can be found at Issue #37