Interbotix/interbotix_ros_arms #35: ee_pose_components.py

sayyidabeegam, 2020-12-09:
Q1.
I am using the code ee_pose_components.py
The input parameters to the code is the end-effector position (x,y,z).
While checking the code, I am not able to find usage of Moveit path planning in the code.
KIndly mention how the .py code is using the ompl path planning algorithm

Q2. I also assume multiple set of joint position (goal pose) of manipulator can reach the same end effector goal position (x,y,z).
The moveit package allows to set the pose(joint poses) and not the goal (x,y,z).
So how is the moveit package taking care of the fact that there multiple goal poses can lead to the  same goal (x,y,x)

swiz23, 2020-12-09:
Hi @sayyidabeegam,

>Q1

That's because MoveIt is not used to do the inverse kinematics or path planning for the scripts in the python_demos directory. Instead, the inverse kinematics solver approach detailed in the [modern_robotics](https://github.com/NxRLab/ModernRobotics) repository is used. About path planning - the joint angles that are calculated by the inverse kinematics solver are just commanded straight to the joints (making sure that no joint limits are violated beforehand). The time it takes for each joint to move, the max speed at which it can move, and the time that each joint spends accelerating are actually registers in the servos themselves that I use to make sure the motion is smooth. However, no collision detecting is done, so you would have to visually make sure that the arm won't collide with anything during movement. In the event of a collision, the servo or servos in collision should realize in a few seconds that they are stalled and torque off - with the LED flashing red, and an Overload error showing up.

>Q2

This assumption is not always the case - especially for arms with fewer than 6 degrees of freedom. But when it is possible for an arm to achieve the same ee position with different joint angles, MoveIt actually *does* sometimes give different joint values. I don't know exactly how MoveIt does this (maybe they give some random joint values as an initial guess). The Python API though used for the python_demo scripts are seeded with three initial guesses - pretty much with all joints at 0 radians except the waist joint which is either 0, -120 or +120 degrees. I've found that these guesses seem to pretty much allow the IK solver to converge every time I've tried (though seeding the function with your own guess is also possible if you're looking for joint values in a specific ballpark area).

The reasons I didn't go the MoveIt approach for the Python scripts are...

- You could just use the Move Group Python API ontop of MoveIt if you want to do this
- Many times, especially for arms with fewer than 6 dof, MoveIt cannot solve the inverse kinematics - so it's not very reliable
- The trajectories that MoveIt comes up with can have variable speed which isn't great if you're trying to do Cartesian motions and want the motion each time to be consistent
- When doing Cartesian motion in the Python API used in the python_demo folder, the initial guess for the IK is the current joint values - this ultimately guarantees that the IK will calculate the same joint values every time for the next goal pose.

As an FYI - just some lingo...
The term 'pose' always refers to the position (x, y, z) and orientation (roll, pitch, yaw, or quaternion) of a given frame (in our case, the end-effector frame) - never joint positions or angles.
MoveIt actually *does* allow you to specify joint space goals or pose goals (see the set_joint_value_target vs set_pose_target functions at .http://docs.ros.org/en/kinetic/api/moveit_commander/html/classmoveit__commander_1_1move__group_1_1MoveGroupCommander.html.

sayyidabeegam, 2020-12-10:
When I try to do move group interface on it , showing error

[ERROR] [1607591874.440900576]: Robot model parameter not found! Did you remap 'robot_description'?
Traceback (most recent call last):
  File "move_to_point.py", line 210, in <module>
    main()
  File "move_to_point.py", line 196, in main
    tutorial = MoveGroupPythonIntefaceTutorial()
  File "move_to_point.py", line 95, in __init__
    robot = moveit_commander.RobotCommander()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/moveit_commander/robot.py", line 151, in __init__
    self._r = _moveit_robot_interface.RobotInterface(robot_description, ns)
RuntimeError: RobotInterfacePython: invalid robot model

I have re mapped the mode as follow
      self.box_name = ''
    self.robot = rx150
    self.scene = scene
    self.move_group = interbotix_arm
    self.display_trajectory_publisher = display_trajectory_publisher
    self.planning_frame = planning_frame
    self.eef_link = x150/gripper_link
    self.group_names = group_names

What else changes needed to do this ?

swiz23, 2020-12-10:
If you're writing your own movegroup Python node, I'd recommend launching it in the [robot_name namespace](https://github.com/Interbotix/interbotix_ros_arms/blob/cb9f5feb6c51190561e05896ba39cd1cce133875/interbotix_examples/interbotix_moveit_interface/launch/moveit_interface.launch#L53). That way, you don't have to specify namespaces directly in the code. You should also remap the two topics specified there so that adding objects works appropriately.

sayyidabeegam, 2020-12-11:
I am using given link as reference http://docs.ros.org/en/kinetic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html  
How can I implement this code to rx150?

closed 2020-12-11 by sayyidabeegam