Interbotix/interbotix_ros_arms #37: Issues with the "use_python_interface" in Moveit

sunnyshi0310, 2021-01-18:
Hi, I want to use the python interface instead of the Rviz in Moveit.  
I tried to run roslaunch interbotix_moveit_interface moveit_interface.launch robot_name:=rx150 use_gazebo:=true use_python_interface:=true moveit_interface_gui:=false
While I didn't see the node "moveit_python_interface" either in "rosnode list" or the "rqt_graph" but still the Rviz interface appears.
Could you help me with that?
Also what we could do to send several commands to the robot using moveit without opening the RViz and set the goals manually. Could we write a python script and publish the goals to some topics (I saw it's "display_planned_path" or something?) and achieve the goals directly?

Thanks in advance.

swiz23, 2021-01-18:
Hi @sunnyshi0310,

To answer your first question...
The moveit_python_interface node was modeled the same way as the official moveit python interface node was as can be seen [here](https://github.com/ros-planning/moveit_tutorials/blob/kinetic-devel/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py). Just like they have the user press the 'Enter' key in the terminal to step through the tutorial, I also do. The launch command you are running automatically launches this node. If you look closely at the the text outputted in the terminal after running the launch command, you should see one that says `"============ Press Enter to begin the tutorial by setting up the moveit_commander (press ctrl-d to exit) ..."`
You might miss it because you will see info about everything else gettting launched at the same time, so you should just scroll through the text to find it.
Anyway, once you press Enter in the same terminal where you launched the launch file, the moveit_python_interface node will be initialized and you will see it show up using the rosnode list command.

About getting rviz not to show up - currently, this is not possible without editing the interbotix_moveit launch file. Just comment out lines 67 - 70 [here](https://github.com/Interbotix/interbotix_ros_arms/blob/cb9f5feb6c51190561e05896ba39cd1cce133875/interbotix_moveit/launch/interbotix_moveit.launch#L67).

About your third question...
Yep! That's the whole point of the tutorial. You can modify (or even better - create your own) python script modeled after the moveit_python_interface node. Just get rid of the 'raw_input' commands, and use the [moveitCommander API](http://docs.ros.org/en/jade/api/moveit_commander/html/classmoveit__commander_1_1move__group_1_1MoveGroupCommander.html) to command your desired pose or joint goals.

A few tips - when you launch your custom python interface node from a launch file, make sure to put it in the 'robot_name' namespace and to remap the attached_collision_object and collision_object topics as shown [here](https://github.com/Interbotix/interbotix_ros_arms/blob/cb9f5feb6c51190561e05896ba39cd1cce133875/interbotix_examples/interbotix_moveit_interface/launch/moveit_interface.launch#L47).

sunnyshi0310, 2021-01-18:
Hi Solomon @swiz23 ,
Thanks for your reply! It looks quite clear now. I will go to try to write my own code, thanks again! Good night!

closed 2021-01-18 by sunnyshi0310

sunnyshi0310, 2021-01-18:
Hi Solomon,

Sorry for bothering again. I try to write my own code and change the "type" of the node in the launch file to the name of my script, but it said it can't locate the node. Do you have any suggestions or maybe I shouldn't do like that.

swiz23, 2021-01-18:
No worries! You should make your script executable. Just navigate to your file in the terminal and type `chmod +x my_file.py` where my_file.py is the name of your file.

sunnyshi0310, 2021-01-18:
Hi, I tried it and also catkin_make the repo. But it still shows the error as: ERROR: cannot launch node of type [interbotix_moveit_interface/moveit_python_test]: can't locate node [moveit_python_test] in package [interbotix_moveit_interface]

swiz23, 2021-01-18:
Is your script located in the https://github.com/Interbotix/interbotix_ros_arms/tree/master/interbotix_examples/interbotix_moveit_interface/scripts directory? Also, if your script ends with '.py', you need to include that as part of the name 'moveit_python_test.py' in the 'type' part of the launch file.