Interbotix/interbotix_ros_arms #38: Shadow Motion

Void12987, 2021-01-19:
Hello Interbotix/Trossen team,

Is there or will there be any support for shadow motion, as seen on https://www.shadowrobot.com/.

I believe there will be a good market for affordable shadow motion.

Thanks,
Nathan

swiz23, 2021-01-19:
Hi Nathan,

I took a look at that website, and it definitely looks pretty awesome. If I'm understanding you correctly, you're asking if we currently have or plan to have some sort of 'glove' a user can wear that can be used to control any of the robot arms?

Currently, we don't have anything like this or plan to have anything like this. The closest thing we have is the PS3/PS4 joystick control feature. However, you could theoretically create something like this yourself and wrap it around code using the Python API to control the robot!

Void12987, 2021-01-19:
Hi Solomon,

Thanks. I very well may try to combine the Interbotix arm API with a VR headset, like https://www.pollen-robotics.com/.

closed 2021-01-19 by Void12987

swiz23, 2021-01-19:
if you do, we'd love to see a video!

Void12987, 2021-01-20:
Yes, if I get to that point, I'll be sure to share it.