Interbotix/interbotix_ros_arms #39: sudo pip install permissions issue

mipark56, 2021-01-20:
Hi there, just following your install wiki and looks good, but found one issue on my system Ubuntu 18.04 + ROS melodic:
In the top-level README, the step
sudo pip install modern_robotics
gives me
  OSError: [Errno 13] Permission denied:
  '/usr/local/lib/python2.7/dist-packages/modern_robotics-1.1.0.dist-info'
A simple fix (had to do this for other projects), was to simply install as user:
sudo pip uninstall modern_robotics
pip install modern_robotics --user

Not a critical fix, but just something that worked for me.  For more info you can see (for example):
https://askubuntu.com/questions/802544/is-sudo-pip-install-still-a-broken-practice

Mike

swiz23, 2021-01-21:
Thanks @mipark56,

I tried running `sudo pip install modern_robotics` on my Ubuntu 18.04 + ROS melodic computer, and did not get any issues.

Just curious - what version of pip were you using (run `pip -V` in a terminal to find this out). Mine is 9.0.1. Also, did you install pip using the `sudo apt install python-pip` command? Maybe you only get that permission denied error on later versions?

In any event, from the article you mentioned - it seems that in general, it's unsafe to use sudo with pip due to the security risk.

mipark56, 2021-01-21:
Hi @swiz23 , thanks for your response.  Here is the output of my pip version:

$ pip -V
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
pip 20.2.3 from /usr/local/lib/python2.7/dist-packages/pip (python 2.7)

Yes, I did run the `sudo apt install python-pip` command without any issues.

If it worked alright for you then perhaps it's not an issue.  As a side note, I have both Python 2 and 3 on my machine which might be (I am not certain) the source of such issues.

closed 2021-01-21 by mipark56

swiz23, 2021-01-21:
Did you upgrade pip (using pip) to get it to version 20.2.3? From my experience, packages installed from apt don't always play nice with packages installed from pip. When you upgrade pip using pip - I think that supersedes the version of pip installed from apt. Then funky things happen...

mipark56, 2021-01-21:
I don't think I upgraded pip (using pip), but that may be possible, I have used this machine for various development for a few months so that could be the case.  I just recall having this issue with another pip package which is why the workaround for me came to me when I saw it.