Interbotix/interbotix_ros_arms #40: Error when connecting the Robot using Virtual Machine

sunnyshi0310, 2021-01-22:
Hi,
I want to use the virtual machine to connect to the physical robot and works on it, while it seems that I cannot open the port. I enable the USB in my VM and when I run the "lsusb", I saw that USB listed there. Do you have any idea about how to solve it? Thanks
[image]

swiz23, 2021-01-24:
Hi @sunnyshi0310,

Using a virtual machine with our robots has not been tested yet. That said, someone in the past has been able to work successfully with our robots using VMWare (as opposed to a VirtualBox). You can see the thread at Issue #29.

sunnyshi0310, 2021-01-25:
Hi,
Thanks, it could work with VMware while another error appears
[TxRxResult] Incorrect status
groupSyncRead getdata failed

swiz23, 2021-01-25:
Alright,

It's possible that since you are using VMWare, there is some delay in the communication between the U2D2 and the computer. Try setting the latency timer to something higher than 1ms (try going up in increments of 5ms). To change the latency timer, modify the file located at /etc/udev/rules.d/10-interbotix-udev.rules such that the `ATTR{device/latency_timer}="X"` part of [this line](https://github.com/Interbotix/interbotix_ros_arms/blob/cb9f5feb6c51190561e05896ba39cd1cce133875/interbotix_sdk/10-interbotix-udev.rules#L13) has an 'X' value of 6ms, and if that doesn't work, 11ms, and if that doesn't work 16ms. Between each change, save the file (you'll need sudo permissions), and unplug/replug the U2D2 into the computer.

sunnyshi0310, 2021-01-25:
Thank you, I will try!

closed 2021-01-25 by sunnyshi0310

swiz23, 2021-04-05:
Just curious - did the suggested fix solve your issue? Or did you get rid of the errors a different way?

sunnyshi0310, 2021-04-06:
Hi Solomon,
I later used a NUC (which has the Linux system) instead. While I could successfully implement the repo in the VirtualBox and the latency error doesn't have an effect on the performance of the arm. Thanks

swiz23, 2021-04-06:
Got it. Thanks for the update!