Interbotix/interbotix_ros_arms #41: Speed Limitation of Rx150 Arm

mluna011, 2021-03-01:
Hello,
I am currently using Interbotix_SDK and 'Time-Based-Control' to move the Rx150 arm to a desired pose. Within the robot_manipulation.py in the Interbotix_SDK package, I am able to change the moving time and acceleration time. Do you know what are the lowest values I can set both of these parameters to, to have the fastest motion of the arm?

swiz23, 2021-03-02:
The lowest setting is 0 for both. However' I'd highly recommend against this as '0' just means that the motors will go their fastest to get from A to B (even though it will of course take more than '0' seconds to move there).
Steps are in increments of 1 millisecond. So next up from 0 would be 0.001, then 0.002, etc...

In general, I'd recommend having each arm motion take at least a second or more. But depending on what your motion is, you could increase or decrease this. For example, if you're trying to move the 'waist' joint 180 degrees, the fastest you'd probably want the motion to be is 1 second. However, if you're just moving the waist joint 45 degrees, you could probably get away with half a second.

mluna011, 2021-03-02:
Oh I see thank you very much!

closed 2021-03-02 by mluna011