Interbotix/interbotix_ros_arms #42: RX200 - Failed to write value[0] on items[Drive_Mode] to [ID : 7]

sayyidabeegam, 2021-03-11:
**We are testing RX200.
But when we try to launch  any file it showing error like that**

[ERROR] [1615462266.610043992]: [DynamixelDriver] Failed to get the Tool
[ERROR] [1615462266.610085329]: Failed to write value[0] on items[Drive_Mode] to [ID : 7]
[ERROR] [1615462266.946160108]: [DynamixelDriver] Failed to get the Tool
[ERROR] [1615462267.035811844]: [TxRxResult] There is no status packet!
[ERROR] [1615462267.035848924]: groupSyncRead getdata failed
[ERROR] [1615462267.035864869]: groupSyncRead getdata failed
[ERROR] [1615462267.035905347]: groupSyncRead getdata failed
[rx200/arm_node-3] process has died [pid 15934, exit code -11, cmd /home/lab_2/interbotix_ws/devel/lib/interbotix_sdk/arm_node __name:=arm_node __log:=/home/lab_2/.ros/log/4386fcbc-825d-11eb-9748-d4bed9c05ca1/rx200-arm_node-3.log].
log file: /home/lab_2/.ros/log/4386fcbc-825d-11eb-9748-d4bed9c05ca1/rx200-arm_node-3*.log
[ERROR] [1615462283.321534911]: Action client not connected: arm_controller/follow_joint_trajectory

**can there be error with hardware, because a little bit heating also there in the last joint motor?**

swiz23, 2021-03-11:
Hi @sayyidabeegam,

The error message that you are getting means that the computer cannot write to the gripper servo (ID: 7). I'm guessing though that is had no issue pinging the servo? Did you got a message that said the following beforehand?
```
ID : 7, Model Number : 1060
```
If so, check to see if the gripper servo is in an error state (red LED is blinking). If that's the case, you'll need to power cycle it for the ROS driver to work. If the error is due to overheating (you can check the Present Temperature register and the Hardware Error Status register using the Dynamixel Wizard 2.0 tool), you'll need to wait for the servo to cool down. Anything aboe 72 degrees Celsius will cause it to go into an error state.

As far as preventing the gripper servo from heating up (I'm assuming you're using the motor in PWM mode), try to avoid having a constant PWM on the servo for more than 30 seconds. Or, you can lower the PWM to something smaller after the gripper is finished gripping an object.

sayyidabeegam, 2021-03-12:
No,I am getting only
**[ INFO] [1615462265.910283404]: ID : 1, Model Number : 1060
[ INFO] [1615462265.945195346]: ID : 2, Model Number : 1020
[ INFO] [1615462265.980189713]: ID : 3, Model Number : 1020
[ INFO] [1615462266.016041864]: ID : 4, Model Number : 1060
[ INFO] [1615462266.055224853]: ID : 5, Model Number : 1060
[ INFO] [1615462266.090155459]: ID : 6, Model Number : 1060**

ID number for 7 is not coming

swiz23, 2021-03-12:
Can you check the three-pin cable going from the wrist_rotate servo to the gripper servo? Is it snugly connected? Does the LED on the gripper servo flash briefly when you power cycle the arm? Is the temperature below 72 degrees Celsius?

sayyidabeegam, 2021-03-12:
The 3 pin wire is connected fine. The LED light comes only once when it is
powered on. Temperature is slightly higher than normal , but it is below 72
degrees.
Is it due to hardware error?

swiz23, 2021-03-12:
Can you post snapshots of all the registers and their values using the DynamixelWizard 2.0 tool?

sayyidabeegam, 2021-03-12:
Sir 
We are not detached the motor from arm yet. Is there any way to confirm without taking motor out?

swiz23, 2021-03-12:
There's no need to detach the motor or remove any wiring to use the Dynamixel Wizard 2.0 tool...

I've attached pictures of what I'm looking for (for the gripper servo - ID: 7) - note that I'm using a WX200 arm in those pics - but it's the same process. You can get the Dynamixel Wizard 2.0 tool [here](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#install-linux).

[image]
[image]

sayyidabeegam, 2021-03-13:
With Dynamixel Wizard 2.0 tool only 6 motors are detecting, last one (7th)
is not coming in list

sayyidabeegam, 2021-03-13:
[image]

swiz23, 2021-03-14:
Did you also try scanning at the 57600 baud rate? That's the default baud rate that comes with the motor (should be set to 1000000 in our office, but it's possible it was not set or it went back to 57600 for other reasons). Also, if the motor doesn't show up in the Wizard, how have you been able to determine what its temperature is?

sayyidabeegam, 2021-03-14:
Yes
We have scanned with all baud rate.
About temperature, we didn't measured the temperature.
When we touch around the gripper motor we feel slightly higher temperatures
than other part of robot. (Only a slight variation)

swiz23, 2021-03-14:
Ok, in that case, you can try to do a firmware recovery on the servo. The hardware setup should be...
1) microusb from your computer to the U2D2
2) three-pin cable from the U2D2 to the motor power hub (this is already done)
3) three-pin cable from the motor power hub to the gripper servo (for this to happen, you first must disconnect the gripper servo from the wrist_rotate servo [ID: 6] as only one servo can be connected during the recovery process; also, you must unplug the three-pin cable going from the motor power hub to the waist servo [ID: 1] for the same reason). There's no need to physically remove the servo - just unplugging/replugging some three-pin wires.

A video showing how to do the recovery can be found at https://youtu.be/PgbIAK2Qg1Y. Note that when it says to turn off/on power, just unplug/replug the 12V power supply to the motor power hub. Also note that the gripper servo model is XL430-W250.

As you've probably guessed, you will either need a three-pin cable long enough to go from the motor power hub to the gripper servo OR you will need to temporarily unscrew the U2D2/motor-power-hub so that you can bring it close enough to the gripper servo to connect it.

If you're not comfortable doing this, you can also ship back the arm to Interbotix so we can fix the issue for you. If you'd like to go that route, please email trsupport@trossenrobotics.com. It would help to include a link to this Issue in your email and mention you spoke to Solomon about it.

sayyidabeegam, 2021-03-15:
> please email ***@***.*** It would help to include a

Okay Sir
We will do that.

swiz23, 2021-03-15:
Sounds good. Another thing to note is that you should reset the gripper's ID to 7 and the baud rate to 1000000 before reconnecting it back to the rest of the arm. By default, after a firmware recovery, the baud rate goes to 57600 and the ID goes to 1.

sayyidabeegam, 2021-03-17:
Sir firmware recovery failed

sayyidabeegam, 2021-03-17:
> On Sun, Mar 14, 2021 at 9:11 AM Solomon Wiznitzer <

sayyidabeegam, 2021-03-17:
[image]

sayyidabeegam, 2021-03-17:
[image]

sayyidabeegam, 2021-03-17:
[image]

swiz23, 2021-03-17:
Looks like your hardware setup is perfect! But the motor firmware you need to install is for the XL430-W250 motor, not the XM430-W350 motor. Can you try with the correct firmware?

sayyidabeegam, 2021-03-17:
Sir 
We have done this because according to the description of RX200 we have three XM430-W350 and four XL430-W250.
But with wizard tool only two is detecting.
So we assumed that one XM430-W350 is wrong.

sayyidabeegam, 2021-03-17:
[image]

sayyidabeegam, 2021-03-17:
[image]

swiz23, 2021-03-17:
That's very strange. The third XM430-W350 motor should be located at the elbow joint (ID: 4). Can you post a snapshot of what your elbow motor looks like? No need to take the arm apart. Just a pic of either side of the motor will do.

sayyidabeegam, 2021-03-17:
The robot we have received has 1 motor lesser than reactor x 200.
But name board is indicating as reactor x 200
Hence, we have received a wrongly assembled  product or a wrong product. The base structure of the robot is also different from what is available on the website.

sayyidabeegam, 2021-03-17:
[image]

sayyidabeegam, 2021-03-17:
[image]

sayyidabeegam, 2021-03-17:
[image]

swiz23, 2021-03-17:
Wow! That is most definitely not the RX200 arm. Looks more like the RX150. I apologize for any inconvenience this has caused you. I'll discuss this with our Operations Manager, Rick, today, and tell you about next steps. Most likely, you'll just need to return the arm, and we'll ship you the correct model.

Can you email me at solomon@trossenrobotics.com so that I can put you in touch with Rick directly?

Again, sorry for the inconvenience!!

In any event, at least we now know why ID: 7 wasn't showing up, considering there was never a 7th motor on your robot to begin with!

sayyidabeegam, 2021-03-17:
Okay Sir

sayyidabeegam, 2021-03-17:
> Hi,
>
> Talked with Rick. Are you the same person who has been emailing E-Store MG
> Super LABS about this issue? If so, Rick has just followed up with the
> store.

Yes Sir
E Store is doing for us.

swiz23, 2021-03-18:
Great! If you have any other questions, just let me know. Otherwise, feel free to close this Issue.

sayyidabeegam, 2021-03-31:
We are informed from e-store that a replacement will be provided for the robot at the earliest.
However, we are not yet provided with a timeline.
By when can we expect to receive the save?

sayyidabeegam, 2021-03-31:
We are informed from e-store that a replacement will be provided for the robot at the earliest.
However, we are not yet provided with a timeline.
By when can we expect to receive the same?

swiz23, 2021-03-31:
Hi @sayyidabeegam,

For the timeline question, please email trsupport@trossenrobotics.com directly. Make sure to include your order number as well.

closed 2021-04-07 by sayyidabeegam