Interbotix/interbotix_ros_arms #43: Running python examples purely in Gazebo

harshitsankhla, 2021-05-03:
Is there a way I can run the python examples purely on a Gazebo spawned arm. It seems the the `interbotix_sdk arm.launch` package expects a physical arm to be connected and the `interbotix_gazebo gazebo.launch` doesn't seem to communicate with the python API. Could be I'm not configuring things right.

Any help would be greatly appreciated.

swiz23, 2021-05-03:
Hi @harshitsankhla,

That is currently not supported, and there are no plans to update this repo with any more features as it's in legacy mode. In general, I'd recommend using our latest arm repository that has all the features of this one and more. See it at https://github.com/Interbotix/interbotix_ros_manipulators/tree/main/interbotix_ros_xsarms.

In the new repo, while you're not able to run the Python scripts in a Gazebo environment, you are able to run the Python scripts on an arm 'simulated' in Rviz. Learn more about it at https://www.youtube.com/watch?v=p0hmgNEqU8Q&ab_channel=Interbotix-TrossenRobotics.

harshitsankhla, 2021-05-04:
Thanks. Moving to the updated repo now.

closed 2021-05-04 by harshitsankhla